shivasiddharth/GassistPi #1: OpenAlsaHandle

garc7481, 2017-05-26:
Hello Im trying to setup my RPi with the instructions you provided. I dont have the AYI Project so I started at the line that saids: **FOR USERS WITHOUT AIY KIT START HERE AFTER SETTING UP AUDIO DEVICE**

I keep getting this error each time I try to run the gassist.sh script:
**OpenAlsaHandle PcmOpen: No such file or directory
[1605:1618:ERROR:audio_input_processor.cc(756)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[1605:1620:ERROR:audio_input_processor.cc(756)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}**

I tried your way cause I wanted the Google Assistant to startup at boot. Before I was able to run it  with no issues. And I was able to send commands and get replies. Any idea why now my I'm getting sound input errors.
Thanks.

Pixelovich, 2017-05-26:
Hello!! 
Great work!
But I have the same issue.
Audio test works, but while launching gassist.sh I faced the same.
Maybe something should be corrected on /boot/config.txt?
I´ll try it.

shivasiddharth, 2017-05-26:
@Pixelovich  and @garc7481 Hi,
 May i know as to what OS you are running ?

garc7481, 2017-05-26:
Thanks for the reply, Im running Raspbian GNU Jessie with PIXEL. As far as my .asoundrc file it looks like this:
**pcm.!default {
  type asym
  playback.pcm {
        type plug
        slave.pcm "hw:0,1"
  }
  capture.pcm {
        type plug
        slave.pcm "hw:1,0"
        }
}**
Like I mentioned before, I had Google Assistant working and running before. I just wanted to make it execute at startup. It looks to me it could be something on the configuration its causing the Assistant to look for a file that doesn't exists. Im trying some things out. I will keep you posted.

garc7481, 2017-05-26:
Just ran the script in DEBUG mode and here is what I get, I looks to me the main.py script has a problem establishing the connection:

**OpenAlsaHandle PcmOpen: No such file or directory
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/connection.py", line 60, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/lib/python3.4/socket.py", line 530, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 345, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 844, in _validate_conn
    conn.connect()
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 284, in connect
    conn = self._new_conn()
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connection.py", line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
requests.packages.urllib3.exceptions.NewConnectionError: <requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x762bab50>: Failed to establish a new connection: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/requests/adapters.py", line 438, in send
    timeout=timeout
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/connectionpool.py", line 649, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.4/site-packages/requests/packages/urllib3/util/retry.py", line 388, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
requests.packages.urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x762bab50>: Failed to establish a new connection: [Errno -2] Name or service not known',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/google/auth/transport/requests.py", line 112, in __call__
    **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py", line 518, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py", line 639, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/adapters.py", line 502, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x762bab50>: Failed to establish a new connection: [Errno -2] Name or service not known',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/src/main.py", line 75, in <module>
    main()
  File "/home/pi/src/main.py", line 69, in main
    with Assistant(credentials) as assistant:
  File "/home/pi/env/lib/python3.4/site-packages/google/assistant/library/assistant.py", line 95, in __init__
    credentials, self._set_credentials)
  File "/home/pi/env/lib/python3.4/site-packages/google/assistant/library/auth_helpers.py", line 40, in __init__
    self._credentials.refresh(http_request)
  File "/home/pi/env/lib/python3.4/site-packages/google/oauth2/credentials.py", line 126, in refresh
    self._client_secret))
  File "/home/pi/env/lib/python3.4/site-packages/google/oauth2/_client.py", line 189, in refresh_grant
    response_data = _token_endpoint_request(request, token_uri, body)
  File "/home/pi/env/lib/python3.4/site-packages/google/oauth2/_client.py", line 104, in _token_endpoint_request
    method='POST', url=token_uri, headers=headers, body=body)
  File "/home/pi/env/lib/python3.4/site-packages/google/auth/transport/requests.py", line 115, in __call__
    raise exceptions.TransportError(exc)
google.auth.exceptions.TransportError: HTTPSConnectionPool(host='accounts.google.com', port=443): Max retries exceeded with url: /o/oauth2/token (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x762bab50>: Failed to establish a new connection: [Errno -2] Name or service not known',))**

garc7481, 2017-05-26:
Ok here is what I found. When I run the Assistant via ENV Terminal, the assistant works perfectly fine. See Bellow:
**pi@mmraspi:~ $ source env/bin/activate
(env) pi@mmraspi:~ $ google-assistant
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'what day is today'}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}**

Now when I run your script using sudo /home/pi/src/gassist.sh, That's when it crashes. I think it has something to do with the (main.py) python script. So audio and .asoundrc are configured correctly. I will keep digging into this.

shivasiddharth, 2017-05-26:
1. Is yours the latest SDK (0.3.0) ?
2. Have you got the python3 env ?

garc7481, 2017-05-26:
Yes I followed your instructions to the letter. Just in case where do I check it.

shivasiddharth, 2017-05-26:
Can you try running the main.py directly ? Using this
`/home/pi/env/bin/python3 -u /home/pi/src/main.py`

garc7481, 2017-05-27:
It works perfect with command. How come the script is not working? what I was trying to accomplish is to have the google-assistant run as a systemd service. But the boot startup mentioned on your instructions doesn't seemed to execute the command propertly and gives the error.

garc7481, 2017-05-27:
Ok. The script works directly from a terminal window, just like the command you stated above:
/home/pi/env/bin/python3 -u /home/pi/src/main.py

But as a script it doesn't work it keeps giving out the original error. Also the script does not startup at boot cause I don't see the process on the task manager.

garc7481, 2017-05-27:
The only way I have managed now to get it working at startup is editing the 
**/home/pi/.config/lxsession/LXDE-pi/autostart**
and then add a line right after @xscreensaver -no-splash
**@lxterminal -e "/home/pi/src/gassist.sh"**

This does execute the script on boot but it keeps a terminal window open on the background of Magic MIrror. But it does works.

Any ideas why your script wont execute using the main.py script

zerocolTHS, 2017-05-27:
I am also ahving this issues 
```
sudo /home/pi/src/gassist.sh
/home/pi/src/main.py:31: RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
GPIO.setup(25, GPIO.OUT)
OpenAlsaHandle PcmOpen: No such file or directory
[2321:2335:ERROR:audio_input_processor.cc(756)] Input error
ON_MUTED_CHANGED:
{'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
{'is_fatal': True}
[2321:2337:ERROR:audio_input_processor.cc(756)] Input error
ON_ASSISTANT_ERROR:
{'is_fatal': True}
```
When i try and run main script 
```
 /home/pi/env/bin/python3 -u /home/pi/src/main.py
/home/pi/src/main.py:31: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(25, GPIO.OUT)
OpenAlsaHandle PcmOpen: No such file or directory
[3831:3844:ERROR:audio_input_processor.cc(756)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[3831:3846:ERROR:audio_input_processor.cc(756)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}

```

AS far as i'm aware i followed your commands to the letter.

shivasiddharth, 2017-05-27:
can you get back to me with the contents of the directory /home/pi/env/lib/python3.4/site-packages/google/assistant

shivasiddharth, 2017-05-27:
@garc7481  @zerocolTHS are you using webcam as microphone ?

zerocolTHS, 2017-05-27:
```
~/env/lib/python3.4/site-packages/google/assistant $ ls
embedded  library  __main__.py  __pycache__

```

Im using a usb microphone. 
Bus 001 Device 006: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller

It has worked before when i first setup google-assistant

garc7481, 2017-05-27:
No webcam. I'm using a USB Mic. I don't think it has nothing to do with the audio or mic settings. I run mine at startup by addding a line to the /home/pi/.config/lxsession/LXDE-pi/autostart.

@lxterminal -e "/home/pi/src/gasssist.sh"

When you reboot google assistant starts and works. It has something to do either the way the main.py script is setup or something on the requests on the script with the GPIO. Not sure but I'm still debugging. Try and run your main.py script with -d at the end of the command to debug.

shivasiddharth, 2017-05-27:
did you try adding the script to /etc/profile ?

garc7481, 2017-05-27:
zerocolTHS I don't think is the audio or mic configuration. When you run it directly from terminal works. Now do you have an LED installed on GPIO25 cause the scripts looks for the GPIO25 to turn on the LED when issuing a command. And turn off when response is finished by the google assistant.

shivasiddharth, 2017-05-27:
This has nothing to do with the GPIOs...  can you try editing the alsa.conf? 
#Edit alsa config file as shown
sudo nano /usr/share/alsa/alsa.conf 
#set the 
defaults.ctl.card 1
defaults.pcm.card 1
defaults.pcm.device 0
defaults.pcm.subdevice 0

shivasiddharth, 2017-05-27:
Just to make sure that the gpios are not causing the issue, can you edit the main.py and remove the gpio codes for leds.

garc7481, 2017-05-27:
I edited the /usr/share/alsa/alsa.conf as per your settings and now I get this error and no response from the google assitant.

**ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'what time is it'}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
OpenAlsaOutputStream PcmOpen: No such file or directory
Start: pcm could not be opened...
[1937:1945:ERROR:audio_output_stream.cc(109)] An error occurred when playing audio output
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}**

shivasiddharth, 2017-05-27:
using rpi's 3.5mm jack for out put? please post the outputs of arecord -l and aplay-l

garc7481, 2017-05-27:
Im using the HDMI output I don't have a 3.5mm jack speaker now. Here are the following:
arecord
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

aplay
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2017-05-27:
Since you are using HDMI, in the alsa.conf, 
change the 
defaults.ctl.card 1
defaults.pcm.card 1
to 
defaults.ctl.card 0
defaults.pcm.card 0

and also your asound.conf should contain the same codes that is there in asoundrc so edit the asound.conf using sudo nano /etc/asound.conf and paste the following contents.

pcm.!default {
type asym
playback.pcm {
type plug
slave.pcm "hw:0,1"
}
capture.pcm {
type plug
slave.pcm "hw:1,0"
}
}

garc7481, 2017-05-27:
Man we are moving back and forth. Thats how I originally had it and it worked. Like I said since the beginning, my audio settings are not the problem cause when i run the google assistant within the terminal window works just perfect.

garc7481, 2017-05-27:
Also questions do you why when you ask the google assistant to tell you the news, she responds here is the news but no audio stream its play.

shivasiddharth, 2017-05-27:
There is no support for news yet !

shivasiddharth, 2017-05-27:
I am trying to re create your error.. Let me see as to whats happening..

zerocolTHS, 2017-05-27:
I edited alsa config file as shown below and all is working now. 
sudo nano /usr/share/alsa/alsa.conf
#set the
defaults.ctl.card 1
defaults.pcm.card 1
defaults.pcm.device 0
defaults.pcm.subdevice 0
------------------------------------------------------------
Over ssh this is the output
 sudo /home/pi/src/gassist.sh
/home/pi/src/main.py:31: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(25, GPIO.OUT)
X11 connection rejected because of wrong authentication.
xcb_connection_has_error() returned true
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'what time is it'}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

Pixelovich, 2017-05-28:
It works!! thanks colleagues. I edited the /usr/share/alsa/alsa.conf file like above.
My only thing to fix is that now i only able to use the mix+speaker of the usb headset in card 1. My integrated 3'5mm jack is not assigned to work now... I think I have to play with this values 1 and 0 in the default.pcms...

shivasiddharth, 2017-05-30:
Since it is an audio config issue closing it.

closed 2017-05-30 by shivasiddharth