shivasiddharth/GassistPi #3: Audio Hierarchy

noiprcocks, 2017-06-10:
Hey,

is it possible to define the "hierarchy" of the usb-soundcard-usage?

For e.g. if I'm already playing Music via spoitfy connect or mopidy the voice gassist can't work and vice versa.

`ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': "how's the weather today"}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
OpenAlsaOutputStream PcmOpen: Device or resource busy
Start: pcm could not be opened...
[1105:1113:ERROR:audio_output_stream.cc(109)] An error occurred when playing audio output
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
`

Is it possible to auomatically stop the Music layback after the hotword so gassist can work?

shivasiddharth, 2017-06-10:
It should be a hierarchy of service. Working on it !

noiprcocks, 2017-06-10:
Awesome, you are!

closed 2017-06-18 by shivasiddharth