shivasiddharth/GassistPi #4: Autostart with CLI on Boot - Problems

noiprcocks, 2017-06-10:
When following this instructions

> #1. Autostart with CLI on Boot

`#TO MAKE IT AUTO START ON BOOT
sudo nano /etc/profile
#IN THE WINDOW SCROLL DOWN TO LAST LINE AND PASTE BELOW
sudo /home/pi/src/gassist.sh
#SAVE AND EXIT BY PRESSING "CTRL+X" AND "Y" AND "ENTER" TO EXIT
#CLOSE TERMINAL`

The Assistant doesnt work/start automatically. It only starts to work automaticaly after I login via putty from a pc.
Theres is a lot on "error Outputs" during boot. See this screenshot:

[http://imgur.com/dncqLJU](url)

Eventually that might help?

Additionally it is no longer possible to start the Pixel Desktop with the "startx" command after i apply the "sudo /home/pi/src/gassist.sh" to "sudo /home/pi/src/gassist.sh"

shivasiddharth, 2017-06-10:
I have clearly mentioned the two ways you can boot in the instruction file. If you want to autostart with the pixel desktop please follow the second method that i have mentioned.

noiprcocks, 2017-06-10:
Hey, yes of course you did:

`clearly mentioned the two ways you can boot`

that is absolutely right.!But it seems like we are talking at cross purposes here. So I'll start once again and try to make the problem more clear.

I am totally aware of the two different options to make the gassist boot on Startup. Like you described.
And I dont want to automatically launch it in the Pixel Desktop.

The headerless cli Version/boot is generally what I want, since the pi will be mainly used in a headless state, build into an old vintage Radio.

But here come the problems:

- when i set it up for cli boot like you described the boot process gives those erros
   http://imgur.com/dncqLJU

- the gassist doesnt start. it only starts after I've logged onto the pi with putty from my Windows pc. i dont have to start it via
`sudo /home/pi/src/gassist.sh`
it just auomatically is on, after i've logged in with user and pw

- it is no longer possible to do "startx" and enter to the Pixel Desktop manually (sometimes desired/needed for easier configuration). after i enter "startx" the Screen stays black and the Desktop doesnt boot. to make the dsktop available again at all via "startx" i have to remove the 
`sudo /home/pi/src/gassist.sh`
from the 
`sudo nano /etc/profile`

What could be causing those Problems? Should autologin eventually be disabled in the 
`preferences-->raspberry pi configurations` ?
Should the
`sudo /home/pi/src/gassist.sh`
and 
`sudo nano /etc/profile`
be done with root?

Hope it could make the Problem more understandable now.

Thanks for your effort, help and Patience.

shivasiddharth, 2017-06-11:
Let me address your concerns one by one.
1. Those errors looks like authentication/network errors. It has nothing to do with the script.
2. This might be a possibility on the first boot after making the advised changes. But on subsequent boots, it should start automatically.

There are so many changes that can be made to the Rpi configs and user settings. It is really hard to test out each and every conditions. All my instructions are based on the assumptions that, they are implemented on a fresh copy of Raspbian without any changes to user/root settings. 

Addressing your issue 3, as i said earlier, if you need the Assistant to auto start on boot in pixel desktop, you need to use the second method.

Hope i made some sense.

closed 2017-06-18 by shivasiddharth