shivasiddharth/GassistPi #5: Cannot start GassistPi

onlize, 2017-06-11:
Hello,

I do not have a kit and followed instructions I have a problem when I have to run this step. Can you help?

(env) pi@raspberrypi:~ $ google-oauthlib-tool --client-secrets /home/pi/assistant.json --scope https://www.googleapis.com/auth/assistant-sdk-prototype --save --headless
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=1039339570934-p5fasdrus16gkugh5cuaqk7gbkik5362.apps.googleusercontent.com&redirect_uri=urn%3Aietf%3Awg%3Aoauth%3A2.0%3Aoob&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&state=q7dbIgrJQ4B1eYLi2g08JCXXZaGloZ&access_type=offline&prompt=consent
Enter the authorization code: 4/e3bwnvewn7t_gvLBWMPDoU1l0Ipwc-C9C4mvN1TTUrg
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line 345, in _make_request
    self._validate_conn(conn)
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line 844, in _validate_conn
    conn.connect()
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connection.py", line 326, in connect
    ssl_context=context)
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/util/ssl_.py", line 325, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.4/ssl.py", line 364, in wrap_socket
    _context=self)
  File "/usr/lib/python3.4/ssl.py", line 577, in __init__
    self.do_handshake()
  File "/usr/lib/python3.4/ssl.py", line 804, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:600)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/requests/adapters.py", line 440, in send
    timeout=timeout
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line 630, in urlopen
    raise SSLError(e)
urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:600)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/bin/google-oauthlib-tool", line 11, in <module>
    sys.exit(main())
  File "/home/pi/env/lib/python3.4/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.4/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.4/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/google_auth_oauthlib/tool/__main__.py", line 106, in main
    creds = flow.run_console()
  File "/home/pi/env/lib/python3.4/site-packages/google_auth_oauthlib/flow.py", line 358, in run_console
    self.fetch_token(code=code)
  File "/home/pi/env/lib/python3.4/site-packages/google_auth_oauthlib/flow.py", line 235, in fetch_token
    **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests_oauthlib/oauth2_session.py", line 221, in fetch_token
    verify=verify, proxies=proxies)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py", line 560, in post
    return self.request('POST', url, data=data, json=json, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests_oauthlib/oauth2_session.py", line 360, in request
    headers=headers, data=data, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py", line 513, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py", line 623, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:600)
(env) pi@raspberrypi:~ $

I tried it twice with the same result. Can you help? What am I doing wrong?
Thank you.

onlize, 2017-06-11:
When I try to start it, I receive the following error:

pi@raspberrypi:~ $ sudo /home/pi/src/gassist.sh
/home/pi/src/main.py:31: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(25, GPIO.OUT)
OpenAlsaHandle PcmOpen: No such file or directory
ON_MUTED_CHANGED:
  {'is_muted': False}[1405:1418:ERROR:audio_input_processor.cc(756)] Input error

ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[1405:1420:ERROR:audio_input_processor.cc(756)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}

I do not have LED connected. Not sure how you want me to do this, should I use resistor? What kind of LED?

In any case, I do not think that LED would cause the problem. When I test my speakers and mic, everything looks good.

Can you help?
Thank you.

shivasiddharth, 2017-06-11:
This has nothing to do with LED. Audio needs to be configured properly. please refer to this closed issue. #1

onlize, 2017-06-11:
Thank you for your reply. I checked that issue and still have the same issue. Interesting that the only way the command sudo /home/pi/src/gassist.sh runs without errors if I change the following in /usr/share/alsa/alsa.conf:
Instead of 
defaults.ctl.card 0
defaults.pcm.card 0
 I use:
defaults.ctl.card 1
defaults.pcm.card 1

In this case, my speakers stop working. If I switch to defaults.pcm.card 0, I have the same error as I specified above.

Can you please help me with this?
Thank you.

shivasiddharth, 2017-06-11:
May i know your audio steup ? As in like, USB mic and 3.5 Jack of Pi or USB DAC that has both Mic and Speaker ports ?

onlize, 2017-06-11:
I have USB Microphone and 3.5 Jack speakers.

`pi@google-assistant:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@google-assistant:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@google-assistant:~$
`
This is my .asoundrc file:
`
pcm.!default {
 type asym
 capture.pcm "mic"
playback.pcm "speaker"
}
pcm.mic {
 type plug
 slave {
  pcm "hw:1,0"
 }
}
pcm.speaker {
 type plug
 slave {
  pcm "hw:0,0"
 }
}
`
All audio tests work for me.

Thank you.

shivasiddharth, 2017-06-11:
Do the following steps
sudo nano /usr/share/alsa/alsa.conf 

Change the default values as folows

defaults.ctl.card 0
defaults.pcm.card 0
defaults.pcm.device 0
defaults.pcm.subdevice 0

#These card values are for USB Mic and Rpi's 3.5mm jack audio out.
sudo nano ~/.asoundrc 

remove the existing code and paste the following

pcm.!default {
type asym
playback.pcm {
type plug
slave.pcm "hw:0,0"
}
capture.pcm {
type plug
slave.pcm "hw:1,0"
 } 
 }

 ctl.!default {
        type hw
        card 0
}

Whatever is there in asoundrc  should be there in asound.conf also. 
sudo nano /etc/asound.conf
pcm.!default {
type asym
playback.pcm {
type plug
slave.pcm "hw:0,0"
}
capture.pcm {
type plug
slave.pcm "hw:1,0"
 } 
 }

 ctl.!default {
        type hw
        card 0
}﻿
save
reboot﻿

onlize, 2017-06-11:
Thank you for your reply. Just wanted to let you know that file /etc/asound.conf does not exist on my computer. I was a few people talking about this file and I tried to reinstall Raspian a few times, but I have never had this file.

In any case, I followed your recommendations and now I receive the following when I try to start it:

ALSA lib conf.c:1697:(snd_config_load1) _toplevel_:18:0:Unexpected end of file
ALSA lib conf.c:3417:(config_file_open) /etc/asound.conf may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3339:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:3788:(snd_config_update_r) hooks failed, removing configuration
OpenAlsaHandle PcmOpen: Invalid argument
[1080:1093:ERROR:audio_input_processor.cc(756)] Input error
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_ASSISTANT_ERROR:
  {'is_fatal': True}
[1080:1095:ERROR:audio_input_processor.cc(756)] Input error
ON_ASSISTANT_ERROR:
  {'is_fatal': True}

Also, when I try to run audio tests, they are failing now:
 speaker-test -t wav

speaker-test 1.0.28

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
WAV file(s)
ALSA lib conf.c:1697:(snd_config_load1) _toplevel_:18:0:Unexpected end of file
ALSA lib conf.c:3417:(config_file_open) /etc/asound.conf may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3339:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:3788:(snd_config_update_r) hooks failed, removing configuration
Playback open error: -22,Invalid argument

arecord --format=S16_LE --duration=5 --rate 16k --file-type=raw out.raw
ALSA lib conf.c:1697:(snd_config_load1) _toplevel_:18:0:Unexpected end of file
ALSA lib conf.c:3417:(config_file_open) /etc/asound.conf may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3339:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:3788:(snd_config_update_r) hooks failed, removing configuration
arecord: main:722: audio open error: Invalid argument

Any ideas?
Thank you very much for working with me.

shivasiddharth, 2017-06-11:
From the aplay -l and arecord -l, i see that you are using a USB Dac which has 3.5mm jacks for both mic and headphone/audio out. Is that right ?

onlize, 2017-06-11:
For testing purposes, I use my headphones with 3.5 jack. There is no mic with the headphones.

I ordered USB mic and it is still somewhere on the way. Temporary I use some professional USB microphone that I borrowed from my brother. Honestly, I do not know what is that mic and if it has speakers. Looks like microphone on stand.

onlize, 2017-06-13:
I received new microphone today and it works now. I guess I had a problem with mic.

I have one additional question. I wanted to do auto start with CLI interface, but it does not work unless I open putty and login. Is this correct?

In any case, it works when I have auto login to Desktop.

Thank you again for all your help.

shivasiddharth, 2017-06-13:
Auto login to CLI should work even without loggining into putty. Just make sure that, it is set to auto login, without password.

shivasiddharth, 2017-06-13:
Since the issue is resolved, i am closing this !

closed 2017-06-13 by shivasiddharth