shivasiddharth/GassistPi #9: The LED pulses all the time, and I need to SSH into the pi to run the system

dthorrington, 2017-08-30:
Thanks for this wonderful run-through on installing the aiyproject system. I had to make a fresh install last night as my SD card corrupted, but this time I needed to install everything headless as I don't have my old HDMI monitor with me at the moment. Your instructions worked a treat, so thank you!

However, a couple of issues:
1 - Once the system is up and running my LED is constantly pulsing. I edited the led.py file to try to resolve this but without success. Your LED does not constantly pulse, so where would I have gone wrong?
2 - The system only works once I SSH into the pi. Previously I only needed to supply power and everything ran without my involvement, save for pressing the LED button. Now I must SSH into the pi for the software to start to look for the hotword. What do I need to do to enable the system once power is supplied, please?

shivasiddharth, 2017-08-30:
@dthorrington , 
First up, I have used the usual Raspbian image and not the AIY image. This implementation that i have documented is based on the Google Assistant SDK and not AIY. So if I am guessing correct, you are now technically running the SDK on an AIY image and hence this blinking issue. It means that, your AIY is running in the background.

Regarding the autostart, try the "Autostart with Pixel Desktop on Boot" method that I have indicated.

dthorrington, 2017-08-30:
Wow, thanks for replying so quickly!

That all makes perfect sense, sorry for not noticing the different images used. I'll do a fresh install using the Raspbian image, thank you!

closed 2017-08-30 by dthorrington