shivasiddharth/GassistPi #15: It's always say's starting snow boy service.

AkterAhsan, 2017-09-20:
I have done all of your instruction. but when I turn on my pi google assistant is ok but sonwboy not working. it's always say's "Starting snowboy service". What can i do. i'm using onboard jack and usb mic.

shivasiddharth, 2017-09-20:
I have set the services to restart if it ran into some error. Did you make any changes to the snowboy.py file ?

AkterAhsan, 2017-09-20:
Nope..

AkterAhsan, 2017-09-20:
Does your ISO file work?

shivasiddharth, 2017-09-20:
I have merged all the projects into one. So old procedures are no longer valid. Did you implement this from scratch by preparing a clean OS ? If you are implementing this over a pre existing assistant it will not work. Snowboy works only with python2.

AkterAhsan, 2017-09-20:
I start with a clean OS

shivasiddharth, 2017-09-20:
run the following command in the terminal and paste the output here.

sudo systemctl status snowboy.service

shivasiddharth, 2017-09-20:
ok, i get your point now. The custom wakeword service has been bundled with Snowboy. So starting snowboy sevice means that the custom wakeword detection is starting. Its not an error.

closed 2017-09-21 by shivasiddharth