shivasiddharth/GassistPi #17: Installing - cannot import name HashMissing

aceycv8, 2017-09-21:
hi,

i've been following your guide to install but when i run script gassist-installer-pi3.sh i get the following

```
pi@SteveAI:~ $ sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh
This script must run as pi, trying to change user...
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_AU
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_AU
Ign http://archive.raspberrypi.org jessie/ui Translation-en_AU
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_AU
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_AU
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_AU
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-dev is already the newest version.
python-virtualenv is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 55 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
libffi-dev is already the newest version.
libssl-dev is already the newest version.
portaudio19-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 55 not upgraded.
Running virtualenv with interpreter /usr/bin/python2
New python executable in env/bin/python2
Not overwriting existing python script env/bin/python (you must use env/bin/python2)
Installing setuptools, pip...done.
/home/pi/env/bin/python2: cannot import name HashMissing; 'pip' is a package and cannot be directly executed
```

can you please assist?

shivasiddharth, 2017-09-21:
Hello,
Are you installing it on a fresh copy of latest Raspbian stretch ?

aceycv8, 2017-09-22:
Hi - this is a fresh copy of Jessie

shivasiddharth, 2017-09-22:
The scripts have been written and tested in Raspbian Stretch, so can you please try the installation in a fresh copy of Raspbian Stretch.

closed 2017-09-22 by shivasiddharth