shivasiddharth/GassistPi #18: Everything works fine...just volume has decreased and the pop sound on wake word is not playing..

AdvaitT17, 2017-09-22:
Everything works fine...just volume has decreased and the pop sound on wake word is not playing..Pls help..

shivasiddharth, 2017-09-22:
Can you mention the board that you are using and the audio setup?

AdvaitT17, 2017-09-23:
I'm using raspberry pi 3 model B and I'm using Sound Card for speaker and mic both...

AdvaitT17, 2017-09-23:
I'm using usb-mic-jack but updated the code:

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 0
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

AdvaitT17, 2017-09-23:
Now when I changed the code to this:(asound.conf)

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:0,0"
        channels 0
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

It gives me pop sound(ding) from my speaker which is attached to sound card and when I ask question it gives answer through my TV speaker connected through HDMI....pLS hELP

AdvaitT17, 2017-09-23:
when I test my speaker through-
speaker-test -t wav 
It works with my speaker attached to sound card and it even works wee will with google assistant service(source env/bin/activate and google-assistant-demo)
It takes tv speakers only when I use custom wake word...

shivasiddharth, 2017-09-23:
1. If you do not hear a particular audio, it has more to do with the frequency response of the speaker that you are using.
2. Just changing hardware ids does not help. hw 0,0 is for jack and hw 0,1 is for hdmi. After changing those hardware ids you need to force the audio. I have clearly mentioned that in the document.

AdvaitT17, 2017-09-23:
What can I do if I'm using an usb card for both playback and capture

shivasiddharth, 2017-09-23:
I had a buffer for the USB DACs previously now i have modfied it. Try the new USB-DAC config.

AdvaitT17, 2017-09-23:
I don't have a USB Dac

shivasiddharth, 2017-09-23:
USB sound card is a USB-DAC. both are one and the same.

AdvaitT17, 2017-09-23:
Ohh... I didn't knew that

AdvaitT17, 2017-09-23:
Now I'm using a usb sound card in 1st jack I insert Speaker jack and in 2nd one I insert mic...Pls suggest

shivasiddharth, 2017-09-23:
please post outputs of aplay -l and arecord -l

AdvaitT17, 2017-09-23:
Now the first problem has been sorted out just the audio volume during answering is low...what can I do for that

shivasiddharth, 2017-09-23:
change the volume in alsa mixer. or it must be the volume of your speaker. I am closing this issue now.

closed 2017-09-23 by shivasiddharth

AdvaitT17, 2017-09-23:
No its not the issue of my speaker and not even alsamixer(Volume is 100%). It happens only during answering and when I tried google-assistant-demo then it works fine...This is problem in wakeword process...