shivasiddharth/GassistPi #19: Help...Everything works fine...just volume has decreased

AdvaitT17, 2017-09-23:
No its not the issue of my speaker and not even alsamixer(Volume is 100%). It happens only during answering and when I tried google-assistant-demo then it works fine...This is problem in wakeword process...

AdvaitT17, 2017-09-23:
In sound config menu shall I change the default option Alsamixer b32... to USB Sound Card for more volume?

AdvaitT17, 2017-09-23:
hello?

shivasiddharth, 2017-09-23:
Alsa mixer has different playback cards to choose from. You need to choose the USB card and set volume to maximum.

AdvaitT17, 2017-09-23:
Ok...I'm trying

AdvaitT17, 2017-09-23:
When I changed it to USB AUDIO DEVICE and rebooted it doesn't gave any response...so I again selected b325... and rebooted so it was working but the response sound was very low...Speaker has no problem because every other thing has sufficient volume even the pop sound on wake word is more than the response give by it while using Assistant

shivasiddharth, 2017-09-23:
share the outputs of aplay -l and arecord -l

shivasiddharth, 2017-09-23:
open the file called assistant.py inside snowboy folder. If you scroll down, you will have a line

self.volume_percentage = 50

change that 50 to 100

AdvaitT17, 2017-09-23:
ok...will I need to reboot

shivasiddharth, 2017-09-23:
Yes !

shivasiddharth, 2017-09-23:
I have made changes to the code now. I have set the volume to 80

AdvaitT17, 2017-09-23:
I changed it but nothing happened

AdvaitT17, 2017-09-23:
I think I have to reinstall snowboy service

AdvaitT17, 2017-09-23:
Its not working yet

shivasiddharth, 2017-09-23:
Reboot the Pi. Now if you use the custom wakeword, google should respond with a higher volume. Just make sure that you have selected the audio card properly.

AdvaitT17, 2017-09-23:
bcm2835 Alsa selected as default in Audio config

shivasiddharth, 2017-09-23:
Is google assistant responding to your custom wakeword ?

AdvaitT17, 2017-09-23:
no its the same as previous one

AdvaitT17, 2017-09-23:
I don't understand...even I put the system volume on mute it still responds with that low volume...How does that happen..

AdvaitT17, 2017-09-23:
Pls help...the volume is very low.

AdvaitT17, 2017-09-23:
This is my asound.conf

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

AdvaitT17, 2017-09-23:
hello?

shivasiddharth, 2017-09-23:
Excuse me. Please learn to use this issue feature. Its not a chat window or a personal messenger and developers need time to look into issues.

AdvaitT17, 2017-09-23:
ok...fine take your own time

AdvaitT17, 2017-09-23:
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

AdvaitT17, 2017-09-23:
arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

shivasiddharth, 2017-09-23:
So this must be your issue. The google assistant's audio is a bit lower than the notification tones. This is because i have amplified all the audio samples  that i have given by 200% so they will be louder than the google assistant's response.

AdvaitT17, 2017-09-23:
Then what can I do

shivasiddharth, 2017-09-23:
In the terminal type alsamixer and then press F6 select your USB card and set the volume levels of the mic and audio out to maximum. The volume that you are controlling in the desktop is the volume of the default bcm card. Hope this clears things up.

AdvaitT17, 2017-09-23:
Shall I make USB AUDIO CARD default setting

AdvaitT17, 2017-09-23:
The voice which was coming before is also gone

shivasiddharth, 2017-09-23:
Sorry i dont understand as to what you mean by "Shall I make USB AUDIO CARD default setting". How do you do that?

AdvaitT17, 2017-09-23:
Preferences>Audio Device Settings

AdvaitT17, 2017-09-23:
Pls help...its my project and I have to display it in the coming week

AdvaitT17, 2017-09-23:
Is there any alternate to fix this problem?

shivasiddharth, 2017-09-23:
For god's sake stop using this issue feature like a chat window. Just edit your comments.
If you change the preferences, the id of the cards will change after a restart in the aplay -l and arecord -l. It will not be 1, it will become 0. Then you need to change the ids in the asound.conf file.

shivasiddharth, 2017-09-23:
Since its not an issue with the project and as it is an issue with your own settings, I am closing this....

shivasiddharth, 2017-09-23:
@AdvaitT17 , All that i can recommend is leave the default as bcm. Change the volume of your DAC in the alsamixer like how i have alredy mentioned above.
"In the terminal type alsamixer and then press F6 select your USB card and set the volume levels of the mic and audio out to maximum. "

Your /etc/asound.conf should have
`pcm.dsnooper {
type dsnoop
ipc_key 816357492
ipc_key_add_uid 0
ipc_perm 0666
slave {
pcm "hw:1,0"
channels 1
}
}
pcm.dmixer {
type dmix
ipc_key 1024
slave {
pcm "hw:1,0"
period_time 0
period_size 1024
buffer_size 8192
rate 44100
}

}
pcm.!default {
type asym
playback.pcm {
type plug
slave.pcm "dmixer"
}
capture.pcm {
type plug
slave.pcm "dsnooper"
}
}`

and paste the same in asoundrc also. 
This will do the job. The key here is to change your DAC volume in alsa mixer.

shivasiddharth, 2017-09-23:
As this has nothing to do with the git and as it is your audio config problem i am closing this issue now.

closed 2017-09-23 by shivasiddharth