shivasiddharth/GassistPi #20: Feature request: Music streaming

aceycv8, 2017-09-24:
Hi,

Is it possible to add music streaming like in?
https://www.hackster.io/vvanhee/internet-streaming-radio-with-google-aiy-1edff3

shivasiddharth, 2017-09-24:
An official update of the SDK is long overdue. I am just waiting to see as to what new features that update brings. I am expecting smart home control and music streaming options to be included in the new update. if google does not update the SDK in the upcoming weeks, i shall include this feature.

ChetanaP76, 2017-10-02:
music streaming and news feature and even google assistant sdk prebuilt features like increase volume, set a timer, set an alarm.

shivasiddharth, 2017-10-08:
**# New Features added:**
**1. Voice control of servo
2. Stream music from YouTube.
3. Indicator lights of GA's listening and speaking events.
4. actions.py script for users to create custom actions.**

Alex9premium, 2017-10-09:
I'm kind of new to this but will it update itself or will I need start by cloning and then doing everything again or just update by using the one-line installer again

shivasiddharth, 2017-10-09:
@Alex9premium , there two updates that we need to address. First update is the updation to the project that i do and second update is the updation that google does to the sdk. 
Unless i do a major overhaul, you can delete the folder and clone the updated project and keep using it. For google's updates, you need to run the one-line installer again.

patts75, 2017-10-10:
Any news on the neopixel connected direct to the GPIO on the PI without Audrino?

Alex9premium, 2017-10-10:
@shivasiddharth I deleted and cloned the repository again but now Google Assistant restart when I say play...

shivasiddharth, 2017-10-10:
@Alex9premium , This updation was a overhaul from last time.. Introduction of any new features means that you need to run the one line installer again.

shivasiddharth, 2017-10-10:
@patts75 , This update already required a good amount of work. Moreover, its just one person working on this, so Neopixel hopefully in the next update..

Alex9premium, 2017-10-10:
@shivasiddharth I tried but got an error at the beginning, maybe it's because I was on headless pi with Google running. I'm not sure how I can close google with no gui from only ssh

shivasiddharth, 2017-10-10:
@Alex9premium , please prepare a new copy of Raspbian stretch and try. infact that will take much lesser time than troubleshooting the existing installation.

Alex9premium, 2017-10-10:
After a complete format and reinstall it works. And can you please make USB MIC and USB SPEAKER output?

ChetanaP76, 2017-10-14:
Music plays....but all the custom commands like How are you, goodnight, remember me to...and all that stuff is not working. sometimes it says "Here are some of the matching videos" and so on...

shivasiddharth, 2017-10-14:
@ChetanaP76 , you mean commands with custom wakeword ? Custom wakewords do not support timers. Please take a look at the google's sdk page.

ChetanaP76, 2017-10-16:
@shivasiddharth Thanks. One more question, can I control my Chromecast through assistant editor or grpc...

shivasiddharth, 2017-10-16:
@ChetanaP76 Sorry,  I dont have a chromecast. So i am not sure.

ChetanaP76, 2017-10-22:
@shivasiddharth can you please add alarms and timers in your next update please

shivasiddharth, 2017-10-22:
@ChetanaP76 , Timers and alarms are already supported by the SDK https://developers.google.com/assistant/sdk/overview

elatrium, 2017-10-23:
Hello Sir,

Somehow i cannot get my new custom wake word (='computer') working.
Just to be sure, to add a new custom wake word from Snowboy I did the following:

1. I have a working setup with the ok-google as wake word.
2. I downloaded the computer.pdml file i created on  Snowboy and
   added this file to the resources directory
3. I change the models configuration in the snowboy.py  
   folder

Is the above sufficient to have the new wake word working?

Thank you,
Sam
Rapsberry Pi 3

shivasiddharth, 2017-10-23:
@elatrium , you need to start the snowboy service.. Details are there in the readme.

elatrium, 2017-10-23:
Thanks. I did start the snowboy service. I presume from your answer above that my actions were correct?
The RP does react on the wakewords Snowboy and Ok Google but not on my created wakeword.
What else could be wrong?

ChetanaP76, 2017-10-24:
@shivasiddharth It only works when I say Ok Google otherwise if I use wakeword it goes till last step that is "your alarm set for 5:30am(for example) sorry something went wrong"
How to do it on gRPC API?

shivasiddharth, 2017-10-24:
@ChetanaP76 gRPC does not support alarms and timers. Its not possible to do something which is not supported. Please see this page.https://developers.google.com/assistant/sdk/overview. Please go though the Google assistant website it has googd amount of information.

shivasiddharth, 2017-10-24:
@elatrium , can you copy and paste the change that you made to the models configuration?

elatrium, 2017-10-24:
I now have it (sort of) working with my wakeword. The problem was that the OK GOOGLE wake word is much more sensitive in responding to my voice then the wakeword downloaded from Snowboy. For my own wakeword (computer) I almost have to 'shout' in the (USB-) microphone and the Google wakeword already reacts even when i whisper. This means it has to do with the wakeword itself. Is there a way to improve the wakeword so it has the same sensitivity as the Google wakeword? Thank you.

shivasiddharth, 2017-10-24:
@elatrium , while training the snowboy wakeword, if there was background noise such a thing tends to happen. While training the wakeword, you can play around with the sensitivity and check as to how sensitive your wakeword is.

ChetanaP76, 2017-10-25:
Can we use custom wakeword without gRPC module

shivasiddharth, 2017-11-11:
@patts75 , I was doing some testing with Neopixels. They seem to interfere with Pi's audio (because Pi's audio works on PWM) and I2S. So in all likely hood, neopixels have to be replaced with some APA102 RGB Leds..

mpskierbg, 2017-11-29:
a top stories feature. I am trying to work it out on my own, but i cant seem to figure it out.

shivasiddharth, 2017-12-01:
@mpskierbg , Am adding it in this update

shivasiddharth, 2017-12-03:
@mpskierbg , i have added the top stories/news rss feed streaming. checkout the git or the video..https://youtu.be/WOyYL46s-q0

rarenu, 2017-12-08:
Dear Mr. Sid, you may consider adding the Status report of Room temperature, Garage door opened or closed etc., by monitoring the ADC input of ESP8266 and convert that text into speech.

dendiee, 2017-12-16:
hello sid , great work on EVERYTHING already! is very cool and usefull.
i thought a feature where after a song is played using ur play functiion it plays the next one based on the current song , like youtube's "next video" feature. this can come in handy when u dont wanna keep saying song after song

shivasiddharth, 2017-12-19:
@tkent42 , I have updated the project (Git) now with actions for controlling Kodi.  Ref: https://youtu.be/ahc6XKYriHo

Alex9premium, 2017-12-19:
Kodi series commands to play the next unwatched episode or a specific episode by saying the name of the season and episode. Thanks

Enzoinventor, 2017-12-23:
after the 3-day sdk update, the ability to specify the language of the assistant's speech with this command has been added:
@ click.option ('- lang', show_default = True,
              metavar = '<language code>',
              default = 'en-US',
              help = 'Language code of the Assistant')
can you add it to your project?
Thanks so much

Enzoinventor, 2017-12-26:
Hi, I saw that you could change the text in question by watching this video: https: //www.youtube.com/watch? v = GMaeY_dmknA
I followed all the steps but I could not make it work. it would be nice to integrate it into your project to get smoother responses compared to pico.

Enzoinventor, 2017-12-26:
I managed to get a decent voice for the speech and I share with you how:
Google’s Text to Speech engine is a little different to Festival and Espeak. Your text is sent to Google’s servers to generate the speech file which is then returned to your Pi and played using mplayer. This means you will need an internet connection for it to work, but the speech quality is superb.

I used used ax206geek’s bash script to access the Google Text to Speech engine (this is an updated version of that script):

Create a file speech.sh with:

nano speech.sh
#the language must be set in speech.sh (tl=en)
Add these lines to the file and save it (in nano editor use CTRL-O writeOut)

#!/bin/bash
say() { local IFS=+;/usr/bin/mplayer -ao alsa -really-quiet -noconsolecontrols "http://translate.google.com/translate_tts?ie=UTF-8&client=tw-ob&q=$*&tl=en"; }
say $*
Add execute permissions to your script with:

chmod u+x speech.sh
Test it using:

./speech.sh Look Dave, I can see you're really upset about this.

just open the action.py file
and modify for example:
say (content)
with
subprocess.call (["./ speech.sh", content)

the code could be improved and suggestions are accepted willingly

Pancer1, 2017-12-30:
Fantastic Project!!! What TTS are You using for notification before comands execution? It sounds a little "robotic". Google TTS is more natural. Is this possible to use Google TTS?

shivasiddharth, 2017-12-30:
@Pancer1 @Enzoinventor , Currently i am using pico2wave, I shall include GTTS in the next update.

Pancer1, 2017-12-31:
Great, thanks! I'm using it in coaxistance with Home Assistant and it's doing great job. 
One more question, how to change hot word? I was searching in main.py and actions.py but didn't find.

Pancer1, 2017-12-31:
One more request. It'll be cool if will be possible to control local alsamixer with voice (local volume up/down). It's usefull if i want control music streaming from YouTube played locally in GassistPi. Now it's only possible to controll Google Assistant and Kodi volume by voice (or i missed it).

shivasiddharth, 2017-12-31:
@Pancer1 , After the update, snowboy was sending the CPU usage of  Pi Zero boards sky rocketing. To avoid confusions, i removed the snowboy completely. When you say volume control of music streaming, it actually means the volume control of VLC (music is played through VLC). In the next update, i will include google music streaming and in the update after, voice control of VLC (that is how i have planned).

tstasevych, 2018-01-03:
@shivasiddharth Thank you for great project. This makes life so much easier for non-developers who like DIY projects. 
I really would love to get that Google Play Music update. Any timetable on that project? Can I sign up to test it?
Also, is there anyway to connect it to [Google Cast SDK](https://developers.google.com/cast/) to allow streaming of music to different chromecast enabled speakers in home. 

Again, thank you for all the time you spend helping others with Google Assistant SDK.

shivasiddharth, 2018-01-03:
@nicusorflorin , In autoplay, the links are dynamically generated. Its quite some some work to implement, i can try, but cannot guarantee implementation.
@tstasevych , its just one person working on this, so there is no ETA for updates.  I shall let you know, once the pre-release branch is available for testing. Google Cast SDK is meant for Android platforms, for linux python, there is something called pychromecast. On two previous occasions, when i started working on the casting, google enabled the casting on the server side, so i am a bit reluctant to spend time on casting feature.

hezzzzz, 2018-01-03:
i want to add led blinking when gassist responds.. and a motor start to run.
should this be an action=? oder is it possible to change the main/actions/putbutton.py? 

thanx

shivasiddharth, 2018-01-05:
@hezzzzz , I have already made provisions for LED indicators. Please go through the README. For controlling stepper motor, you can use the function that i have already created in the actions.py file.
@nicusorflorin, Managed to get autoplay working. Will release it in the next update.

hezzzzz, 2018-01-05:
@shivasiddharth:
LEDs: yes i know but i need the leds to be always on and just blink when gassist listens/responds..

thanx

nicusorflorin, 2018-01-06:
@shivasiddharth awesome, you are the best!

shivasiddharth, 2018-01-07:
@hezzzzz , Even that provision is there, I have used PWM to blink LED connected to GPIO25, which turns off at the end of conversation. Instead of turning off using led.ChangeDutyCycle(0), use led.ChangeDutyCycle(100) to always have it turned on.

shivasiddharth, 2018-01-09:
@tstasevych  and @nicusorflorin , Project has been updated with google music streaming and youtube autoplay features.. Update and enjoy !

Pancer1, 2018-01-09:
YouTube autoplay? What does this feature really do?  What has changed?

shivasiddharth, 2018-01-09:
Youtube autoplay will play the next video/track that comes up in youtube suggestions much like how is in the web. Got totally immersed with Google Music that I forgot about GTTS, will add it in the next update.

nicusorflorin, 2018-01-10:
@shivasiddharth thank will get to it once i get home, i will have to see how i can manage the update without losing my own humble customization :) (any suggestion how best to go about it? should i just merge the master branch into my local one?) (added a few extra actions)

shivasiddharth, 2018-01-10:
@nicusorflorin , Use the update script that i have given at the top of README. I know that some of you might have tweaked some actions, so that script will rename/backup your existing GassistPi folder and then clone the updated git. Once all the installation and updation is done, its as easy as copying your customizations from the backedup folder to this new one.

11mail22, 2018-01-17:
missing snowboy support as offline hotword detection, as in the past. Will it come together with new SDK?
Language switch support? (aiy.i18n.set_language_code)

VinceW31, 2018-01-19:
Fantastic project, many thanks for all the hard work! much better than Alexa.
Can you please look at implementing multiple user profiles as now available in Google Home? 
This would allow my wife to fully use the device as well as me, at the moment I can access my own calendar events and personal lists, but other members of the family cant get that functionality (unless Im missing something).  Cheers! and keep up the fantastic work, we all really appreciate it :-)

Pancer1, 2018-01-25:
One more idea! Maybe make a possibility to have a file whjere user could write down his own JSON requests with address to send every one of them? For example using CURL ?
I'im using GassistPi with Home Assistant. Now I'm using IFTTT for integration with some other systems. It will be cool to use GassistPi intstead. IFTTT has big delay (requests are processed in the cloud). Do You know what i mean?

VinceW31, 2018-01-27:
I run my RasPi3 with modified GassistPi code controlling multiple SOnOff wireless switches and ESP8266 devices, this gives me a customisable and extendable Voice Controlled Home Automation setup.  I have modified the GassistPi code so it supports multiple SOnOff wireless switches and ESP8266 using simple http requests, in this way it does not use IFTTT at all, the http requests go directly from the RasPi across my own LAN to control the wireless devices so there is no real delay.   Ive also changed the way the code detects a switch request for a specific device, it no longer needs a “wireless” or a “trigger” command, it just  looks for a unique device or appliance name, for example “Lamp”, “Dishwasher”, “Washing Machine”, “Lounge Lights” etc.    When it detects one of the listed device/appliance names then it looks for the respective command like “On”, or “ Off” etc.  This makes it much easier for the user (or wife!) as they can use a more natural  phrase or string of words like from a simple “”Lamp On” or “Switch the Lamp On”, to something like “Please turn on the nice red lamp next to the setee” and they all will work ok as the initial trigger is the word “Lamp” in this example.
 
Cheers, Vince

Pancer1, 2018-01-27:
I think this is what I need. Could you share your code? I'm not a programmer so it's hard to write something from scratch. I'll try to edit your code for my project.

[16 more replies]