shivasiddharth/GassistPi #21: Assistant responds even if the wakeword is not called

AdvaitT17, 2017-09-25:
Assistant responds even if the wakeword is not called...Even if I say hello it detects.

shivasiddharth, 2017-09-25:
This is called false positive detections. The some models that i have given are generalised and some personal models. The pronunciation differs from person to person, so for more accurate detection, you should train the snowboy with your voice and then download the model. You can find more details about this [here](https://snowboy.kitt.ai)

AdvaitT17, 2017-09-25:
I have dowbload my own model only but then also it identifies some other thing....like it cold be anything not only hello but everything...What should I do for that

shivasiddharth, 2017-09-25:
If you recorded the model in a noisy environment it will tend to happen like that. There should be absolute silence when recording the models. This is an issue that you should open in snowboy git. Not here.

closed 2017-09-25 by shivasiddharth