shivasiddharth/GassistPi #22: Change trigger word from "OK Google" to something we want

tkamaal, 2017-09-26:
Hello Sid,

**I wanted to change the trigger from "OK Google" to something else without using GassistPi. Please provide some information that would me do it.** 

I have Raspberry Pi 3 running NOOBs OS. So far, I have installed google-assistant-sdk and have it running successfully. Voice trigger with command "OK Google" works fine.

shivasiddharth, 2017-09-26:
As of today, custom wakeword without 3rd party apps like snowboy or pocketsphinx is not possible. The assistant API does not have an option to change the hotword or wakeword. In the upcoming months (don't know when), google may release an update with an option to change the hotword, untill then you may have to use snowboy or other similar apps.

tkamaal, 2017-09-26:
Thank you for clarifying @shivasiddharth. Will start using GassistPi then.

closed 2017-09-26 by tkamaal