shivasiddharth/GassistPi #24: Replaced main.py is not working

helloworld2k, 2017-09-27:
Hi Shiva, I tried replace the main.py from folder 'Neopixel Indicator' to src and reboot, but the Google assistant is not working anymore (even no boot up voice), and I replaced back to the original main.py, it works again. do you know anything wrong with this?

shivasiddharth, 2017-09-27:
What change did you make ? Manually run the assistant and post the error.

helloworld2k, 2017-09-28:
I just added time.sleep(1)  into the code, I don't understand the problem, but after a few hour trying, it is ok now, thank you.

BTW, I cannot wake up the GAssistant by "snowboy" or "alexa" on my Pi3, everything I just followed the README

shivasiddharth, 2017-09-28:
Interesting... Where did you add the sleep code ?

closed 2017-09-28 by shivasiddharth