shivasiddharth/GassistPi #27: Like in google we say ok google stop we can't do it with custom wakeword

ChetanaP76, 2017-10-02:
Like in google we say ok google stop we can't do it with custom wakeword. When I was testing an application it was not stopping when i say my custom "wakeword stop" like ok google stop

shivasiddharth, 2017-10-02:
Custom wakeword is based on the gRPC modules. Not all features of the assistant library are supported by the gRPC.

ChetanaP76, 2017-10-02:
can we add mods to them like ktinkere.co.uk as  in gRPC modules don't have action.py

shivasiddharth, 2017-10-02:
Please take a look into the main.py in src folder and assistant.py in snowboy folder. You can make similar mods for the projects in the website that you shared. If you still need an action.py file, add your request here #20

ChetanaP76, 2017-10-02:
Thanks shivasiddaharth you are doing a great job

shivasiddharth, 2017-10-08:
I have created an actions file, but copy pasting AIY actions will not work. for a better understanding of the new actions file please follow this video. https://youtu.be/-MmxWWgceCg

closed 2017-10-08 by shivasiddharth