shivasiddharth/GassistPi #28: google assistant service is responding before snowboy

moham96, 2017-10-06:
Hi,
I'm running Gassistpi on raspberrypi 3, according to the readme file I should run snowboy and google assistant service at the same time but sometimes google assistant takes the lead and respond before snowboy, for example when i say "Alexa" google assistant responds "I'm flattered but that's not me, I'm your google assistant" i don't know how to reproduce this issue, it happens randomly, this happened three times for me already

shivasiddharth, 2017-10-07:
That will happen if you trigger google assistant and then say "Alexa" as a query. Are you using HDMI out for audio?

moham96, 2017-10-07:
@shivasiddharth I haven't setup any other mean to trigger google assistant(gpio button) other than the voice activation, and no i'm not using hdmi out, i'm using audio jack for speakers and usb mic

shivasiddharth, 2017-10-07:
Please understand, Ok-Google does not respond to alexa. What i am trying to say is, ALexa would have triggered the assistant and saying alexa again will give "I'm flattered but that's not me, I'm your google assistant"

moham96, 2017-10-07:
@shivasiddharth ok I understand now ,you mean i might have said "Alexa" two times quickly, thank you for clearification

closed 2017-10-08 by shivasiddharth