shivasiddharth/GassistPi #34: Gpio controls aren't working...

PavanSN, 2017-10-13:
I said ok,Google and said the command many times but unable to toggle the gpio.. plz help.

shivasiddharth, 2017-10-13:
What response did you get ?

PavanSN, 2017-10-13:
It just says, I'm working on it...

shivasiddharth, 2017-10-13:
it means, its not recognizing your trigger word.. Start the assistant manually from terminal say the command for triggering and see what it is detecting. the detected text will appear as test in the terminal.

PavanSN, 2017-10-15:
It detected the word and said I can't help you at the moment... Plz help

shivasiddharth, 2017-10-15:
google has some issues with their servers Please take a look at this
https://developers.google.com/assistant/sdk/release-notes
googlesamples/assistant-sdk-python#119

shivasiddharth, 2017-10-15:
Even if the demo recognizes the trigger word it cannot execute the actions..

closed 2017-10-15 by shivasiddharth