shivasiddharth/GassistPi #38: google-assistant-demo error

tony1550, 2017-10-18:
I followed the instructions very carefully and I was able to get into the (env) and then the command 
google-assistant-demo   gives me an error that reads: 

usage: google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
google-assistant-demo: error: argument --credentials: No such file or directory: "/home/pi/.config/google-oauthlib-tool/credentials.json"

I believe that it is looking for the credentials file, but after I authenticated and entered the code it said that is was saved in 

/root/.config/google-oauthlib-tool/credentials.json

can anyone tell me how to fix this?

shivasiddharth, 2017-10-18:
Can you confirm that you have named your credentials file as assistant.json ? The authorization issue sometimes arises due to overwriting of files already existing in the environment. Try running the one-line installer again.

tony1550, 2017-10-18:
Yes, I did rename the credentials file to "assistant.json" and it's currently in my /home/pi/ directory. I will try to re-run the installer.

closed 2017-10-23 by shivasiddharth