shivasiddharth/GassistPi #39: custom wake word is not working properly.

AkterAhsan, 2017-10-19:
Hey Sid, custom wake word is not working properly. with ok google, everything works perfect but with custom wake word it sounds like so destructible. if I ask something it detects something else.
I don't want multiple custom wake word, I need one custom wake word with GPIO and no freaking out. like your previous video: https://www.youtube.com/watch?v=QO3CCScg9-s&t=77s 

please help me with that, how I can make one custom wake word and change the trigger word.

AkterAhsan, 2017-10-19:
I noticed when I give the same command to "Snowboy" wakeword and "Ok Google" wakeword, the assistant gave a different answer. For example, I give a command "Let's go!". Ok Google will say "I'm ready!", whilst snow will answer either "sorry, I can't help with that yet" or "here's matching videos". The same thing if I give a command such as "Can we go now?" "Can you hear me now?"

shivasiddharth, 2017-10-19:
google has some issues with their servers. Please take a look at this
https://developers.google.com/assistant/sdk/release-notes
googlesamples/assistant-sdk-python#119

closed 2017-10-19 by shivasiddharth