shivasiddharth/GassistPi #40: google-assistant-demo Not found

lawrence-jeff, 2017-10-20:
Have ran through the instructions a few times, running the instructions in step 7 result in command not found:

pi@raspberrypi:~ $ source env/bin/activate  
(env) pi@raspberrypi:~ $ google-assistant-demo 
bash: google-assistant-demo: command not found
(env) pi@raspberrypi:~ $

shivasiddharth, 2017-10-20:
May i know as to what board you are using? And also the installer that you used ?

lawrence-jeff, 2017-10-21:
Sorry should have been more detailed. I am using a pi zero wireless
Initially I started with Jessie (using Pi Bakery). I noticed the issue there, I then moved to using the latest Stretch zip on the official site (I believe its from September)

Followed the instructions I believe to the tee, the other pushtotalk demo ends up there and mostly working but the google-assistant-demo is always not found.

shivasiddharth, 2017-10-21:
google-assistant-demo is the "Ok-google' hotword version of SDK. It does not work on Pi Zero W. That is why, Point 7 says, Move into the environment and test the google assistant **according to your board**

lawrence-jeff, 2017-10-21:
Ok -I assumed 'according to your board' meant more about how your hardware is setup (more about mic config). Since the previous steps all differentiate between Pi zero and Pi 2/3 more clearly, maybe changing this to say 
Move into the environment and test the google assistant according to your board
source env/bin/activate  

For Pi Zero (Doesn't support hotword detection)
googlesamples-assistant-pushtotalk   

For Pi 2/3
google-assistant-demo 
googlesamples-assistant-pushtotalk   

Probably clear to people that are familiar with the SDK - but for someone just following the instructions and assuming the boards have the same functionality it was a bit unclear. Thanks for you work with this.

shivasiddharth, 2017-10-21:
Thanks for your suggestion, will make that change..

lawrence-jeff, 2017-10-21:
Didn't want to open another issue in case it was something silly on my side, but Is there some trick to getting a pi zero to work.

Followed the instructions and can get the pusthtotalk sample to work correctly (using a USB headset for mic and audio out).

Setup the headless services per the instructions and when I reboot I hear the announcement that I can now use custom wakewords - but if I do "Hey google" or "OK google" I get no response, I don't hear anything else until I reboot the device and I get a new announcement.

I can't seem to find any log being generated even though I see a logger class in the code so not sure where to look.

Any guidance would be appreciated

Thanks

Jeff Lawrence

Closed #40<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fshivasiddharth%2FGassistPi%2Fissues%2F40&data=02%7C01%7Clawrence_jeff%40hotmail.com%7C126ed8ba7f9f4d81ff3508d51891a9c6%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636441936245743827&sdata=tuPHOOrKBhDHoSdFopH3BYraI5MV%2BdBcnSSSPLxlsKY%3D&reserved=0>.

lawrence-jeff, 2017-10-21:
While talking to my alexa device with my headset still on I  accidentally discovered the code seems to trigger off 'alexa' not anything google related, which seems odd ..

So I can now get a beep followed by another audio sound but it doesn't respond with an answer after I ask a question. After 1 or 2 requests it just stops responding to 'alexa' as well.

Really hoping to get this to work as the Zero Ws are $5 dollars now at Microcenter

Jeff

________________________________
From: Jeff Lawrence <Lawrence_Jeff@hotmail.com>
Sent: Saturday, October 21, 2017 12:34 PM
To: shivasiddharth/GassistPi
Subject: Re: [shivasiddharth/GassistPi] google-assistant-demo Not found (#40)

Didn't want to open another issue in case it was something silly on my side, but Is there some trick to getting a pi zero to work.

Followed the instructions and can get the pusthtotalk sample to work correctly (using a USB headset for mic and audio out).

Setup the headless services per the instructions and when I reboot I hear the announcement that I can now use custom wakewords - but if I do "Hey google" or "OK google" I get no response, I don't hear anything else until I reboot the device and I get a new announcement.

I can't seem to find any log being generated even though I see a logger class in the code so not sure where to look.

Any guidance would be appreciated

Thanks

Jeff Lawrence

shivasiddharth, 2017-10-21:
The custom wakewords that i have set by default are Alexa and Snowboy. They will not respond to Hey google or ok-google. Inside the resources folder, i have given couple of other voice models, which you can use, but they are personal voice models trained by my voice and may not work effeciently for others. For changing the voicemodels, open the snowboy.py file and scroll to a line that says models and change the name of the model there. The names should match the ones given in the resources directory.

The custom wakeword uses gRPC API and right now google has some server issues with that. Please refer to this link https://developers.google.com/assistant/sdk/release-notes for more information.

lawrence-jeff, 2017-10-21:
Ok - that helps, not sure if it is the google service but its not usable at this point on a Zero.

If I fresh boot it I can get it to respond once or twice to simple things like Alexa - What time is it. But if I ask it the forecast I here an "It" which then cuts off ,  After that it seems the service has crashed, it will no longer respond to the wakeword

Also not sure if it matters but Snowboy didn't work for me as a hotword out of the box, only alexa.

The only other thing I have running is VNC - would that potentially break it? I can disable and try again

Thanks for the help with it though.

Jeff

shivasiddharth, 2017-10-21:
Turning off vnc might have marginal impact. It has a lot to do with the lower processing power of the Pi Zero. There is really nothing much that can be done.

closed 2017-10-23 by shivasiddharth