shivasiddharth/GassistPi #42: Issue while playing music

santoshmohan, 2017-10-24:
While playing music, I am facing issues:
```def YouTube(phrase):
      print(phrase)
      idx=phrase.find('play')
      print(idx)
      track=phrase[idx:]
      track=track.replace("'}", "",1)
      track = track.replace('play','',1)
      track=track.strip()
      print(track)
      global playshell
      if (playshell == None):
        playshell = subprocess.Popen(["/usr/local/bin/mpsyt",""],stdin=subprocess.PIPE ,stdout=subproce$
      print("Playing: " + track)
      playshell.stdin.write(bytes('/' + track + '\n'))
    playshell.stdin.write(bytes('1' + '\n')) 
    playshell.stdin.flush()

```
If pharse is "Play shape of you", it fails while stdin.write and below is the exception

```
shape of you
Playing: shape of you
Traceback (most recent call last):
  File "/home/pi/main.py", line 76, in <module>
    main()
  File "/home/pi/main.py", line 71, in main
    YouTube(str(usrcmd).lower())
  File "/home/pi/actions.py", line 51, in YouTube
    playshell.stdin.write(bytes('/' + track + '\n'))
TypeError: string argument without an encoding
pi@raspberrypi:~ $ Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
BrokenPipeError: [Errno 32] Broken pipe
```

shivasiddharth, 2017-10-24:
Any reason for adding `playshell.stdin.write(bytes('1' + '\n')) `  in the youtube function ?

santoshmohan, 2017-10-24:
As ` playshell.stdin.write(bytes('/' + track + '\n1\n'))` thing wasn't working, I had to split them in to two.
But still it didn't work

shivasiddharth, 2017-10-25:
I tested the  "Play shape of you" command and i can confirm that it works. Please remove the changes that you made and make sure that the code is formated properly and try again. Sometimes due to slow internet connection, i might take some time.

santoshmohan, 2017-10-28:
[image]
Still I am facing the same issue. I am using same code.

shivasiddharth, 2017-10-28:
What is "the same code" ? Your edited version or the unedited version? It will fetch only the best and topmost rest from YouTube search. It's not Google to fetch some random song. If it did not find a matching result it will return an error.. some song is too vague query.

santoshmohan, 2017-10-28:
Issue resolved by changing the below:
`playshell.stdin.write(bytes('/' + track + '\n1\n','utf-8'))`

closed 2017-10-28 by santoshmohan