shivasiddharth/GassistPi #47: Custom commands issue

totheM, 2017-10-27:
I have problems adding custom commands. 

In the action.py i added a simple command like 

```
def party():
    os.system("./codesend 4289795")
```

I tried adding in assistant.py  and main.py and still not working

**`from actions import party`**
 

I have watched your video but no idea how to make this working

The only way i made it work was to edit **stop** like below and only stop command was working, any other custom command that i'm adding it's not working, please guide me

`def stop():
    pkill = subprocess.Popen(["/usr/bin/pkill","vlc"],stdin=subprocess.PIPE)`

to 

```
def stop():
    os.system("./codesend 4289795")
```

I tried different formats with no success 

```
def party():
    subprocess.Popen(["./codesend 4289795"], stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
    os.system("./codesend 4289795")
    subprocess.call(["/codesend 4289795"], shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
```

Thanks

shivasiddharth, 2017-10-27:
After importing, you need to create a trigger in the main.py else it will not work. Take a look at lines 95-102 in the main.py

shivasiddharth, 2017-11-13:
Closing this...

closed 2017-11-13 by shivasiddharth