shivasiddharth/GassistPi #48: How to get rid of the tone heard after 'Hey google'?

kbear99, 2017-10-28:
Sid, first of all thanks so much for all the hard work.  I was looking for a way to shutdown the Pi using a voice command since my Assistant setup will not have keyboard/monitor.  This is perfect for that.

With this AIY implementation however, there is a tone (similar to the hangout notification) that is played every time after I say the wake word, which I find quite annoying.  Usually I can ignore the tone and speak my request over it, but it can still be disruptive and sometimes seems to cause the Pi to lag for a moment.  Is there any way to get rid of this tone?

BTW, prior to step 7 of your instructions for testing, I found it necessary to reboot the Pi first before it would work properly.  You might want to add this step to your instructions.

shivasiddharth, 2017-10-28:
Open the main.py file inside the src folder and scroll to a line that reads,
`subprocess.Popen(["aplay", "/home/pi/GassistPi/sample-audio-files/Fb.wav"], stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE) `
and add a # before the start of that line.

kbear99, 2017-10-29:
That did the trick.  I have much to learn.  Thanks kindly.

closed 2017-10-29 by kbear99