shivasiddharth/GassistPi #49: Music stream ends abruptly

theboss1956, 2017-10-28:
I have been happy with the project so far, but whenever I play music, it plays a little more than halfway through, and then stops. Running on RPI3. Any solutions?

shivasiddharth, 2017-10-28:
its an inherent issue with the youtube downloader and vlc player..refer https://github.com/mps-youtube/mps-youtube/issues/543
i need some time to work on the fix. changing player to mpv doesn't seem to work..

theboss1956, 2017-10-29:
Can you update me when it’s fixed?

shivasiddharth, 2017-11-05:
Hello,
This is let you know that i have fixed the abrupt stoppage issue today. I tested the playback and almost music played smoothly for about 2 hours. If you try the installation again, it should be fine..

theboss1956, 2017-11-05:
Thanks! Will close

closed 2017-11-05 by theboss1956