shivasiddharth/GassistPi #51: assistant is triggered by almost any word

moham96, 2017-11-04:
Hi,
I'm running google assistant and snowboy on  a raspberrypi 3 using a usb mic, but although the snowboy.py is using only two models (alexa.umdl and snowboy.umdl) google asistant is triggered by alot of words, some of the words that i tried:
-cortana
-siri
-android
-mark
even some arabic words trigger google assistant and sometimes when i'm talking on the phone or with someone next to the mic, this happens mostly if i'm away from the mic about 1 to 2 meters,I don't know what s causing this but could it be caused by a cheap mic ?
even some arabic words trigger google assistant and sometimes when i'm talking on the phone or with someone next to the mic, this happens mostly if i'm away from the mic about 1 to 2 meters,I don't know what's causing this but could it be caused by a cheap mic ?

shivasiddharth, 2017-11-04:
Reduce the mic volume, do not set it to 100. anything between 60-70 is ideal. Even after that, if your mic is picking up ambient noises, reduce the sensitivity in the snowboy.py file.

shivasiddharth, 2017-11-27:
Closing this due to inactivity...

closed 2017-11-27 by shivasiddharth