shivasiddharth/GassistPi #52: im having issues to import library

Djancyp, 2017-11-04:
I'm trying to add a library for ex: gtts but main.py doesn't work.

shivasiddharth, 2017-11-04:
Can you provide more details like the code that you have modified and explain a bit more as to what you are trying to do. This just does not provide enough details for me to help.

Djancyp, 2017-11-04:
import os
import random
import pygame
from gtts import gTTS
import keyboard
import subprocess
pygame.init()
pygame.mixer.init

def speech(sp):
	speech = "speech/"+ sp +".mp3"
	if os.path.isfile(speech):
		os.system("mpg321 "+ speech)
		os.system("clear")
	else:
		tts = gTTS(text=sp, lang='en')
		tts.save("speech/" + sp +".mp3")
		
		os.system("mpg321 "+ speech)
	

def findmp3():
    for root, dirs, files in os.walk('/home/djan/Music/'):
        for filename in files:
            if os.path.splitext(filename)[1] == ".mp3":
                yield os.path.join(root, filename)

def randomsong():
	list1 = songs()
	song = random.choice(list1)
	return song

def songs():
	list1 =[]
	for mp3file in findmp3():
		list1.append(mp3file)
	return list1

def play():	
	s = 'playing-song'
	speech(s)
	song = randomsong()
	pygame.mixer.music.load(song)
	pygame.mixer.music.play()

def stop():
	s = 'stop'
	speech(s)
	pygame.mixer.music.stop()
	 
def nextsong():
	s = 'next-song'
	speech(s)
	next = randomsong()
	pygame.mixer.music.load(next)
	pygame.mixer.music.play()
def pause():
	s = 'pause'
	speech(s)
	pygame.mixer.music.pause()
def unpause():
	s = 'play'
	speech(s)
	pygame.mixer.music.unpause()

if __name__ == "__main__":
	
	keyboard.add_hotkey('p', play)
	keyboard.add_hotkey('t', pause)
	keyboard.add_hotkey('y', unpause)
	keyboard.add_hotkey('s', stop)
	keyboard.add_hotkey('n', nextsong)

	keyboard.wait()

Djancyp, 2017-11-04:
I wrote this code and after i import ill add trigger to start playing music but I cant import anything.

Djancyp, 2017-11-04:
I've tried to save this file separately and import like "from playlistpi import *" but won't work also I've  tried to import just from gtts import gTTS its not working nighter.  Can you help me with that, please?

shivasiddharth, 2017-11-04:
From what i see from the code, you are trying to create a mp3 player which give a voice output for different actions like playing or pausing etc. I dont understand as to how the google assistant is even involved with this or the main.py. For such general queries please post your questions in the raspberrypi.org forum, where someone might be able to help.

closed 2017-11-04 by shivasiddharth