shivasiddharth/GassistPi #55: Youtube Playback

valascus, 2017-11-06:
Raspberry Says: Playing $music but nothing is played.

shivasiddharth, 2017-11-06:
open a terminal and enter he following
`mpsyt set player chromium-browser, set playerargs ,exit`

close the terminal and try again

valascus, 2017-11-07:
Nope.
I tried to play some music directly at mpsyt. 
Says only: error. retrying

valascus, 2017-11-07:
Solved it. Didn't have mpv installed. Now, google assistant can't stop it.
Even pkill chromium can't stop it.

// Find out vlc is the one gassistpi uses, but can't get anything out of it. Chromium-browser was also not installed, but, even after installing it, it was not working. Only solved by installing mpv and setting it as player

shivasiddharth, 2017-11-07:
Normally when you update and upgrade the system, chromium will be installed, but to play it safe, i shall include it in the installer script now.  mpv has to be built and installed. That is why i have gone with with chromium instead. Since the issue is resolved, i am closing this.

closed 2017-11-07 by shivasiddharth

valascus, 2017-11-07:
probably it would't install, because i'm using the lite version of raspbian.
Thanks!