shivasiddharth/GassistPi #57: Radio Station - Can't download?

valascus, 2017-11-07:
Nov 07 03:20:12 raspberrypi python[1460]: [01a98de0] [cli] lua interface: Listening on host "*console".
Nov 07 03:20:12 raspberrypi python[1460]: [cli] lua interface: Listening on host "*console".
Nov 07 03:20:12 raspberrypi python[1460]: [71700a10] filesystem access error: cannot open file /home/pi/p (No such file or directory
Nov 07 03:20:12 raspberrypi python[1460]: [71700a10] core access error: File reading failed
Nov 07 03:20:12 raspberrypi python[1460]: [71700a10] core access error: VLC could not open the file "/home/pi/p" (No such file or di
Nov 07 03:20:12 raspberrypi python[1460]: [71c00510] core input error: open of `file:///home/pi/p' failed
Nov 07 03:20:12 raspberrypi python[1460]: [71c00510] core input error: Your input can't be opened
Nov 07 03:20:12 raspberrypi python[1460]: [71c00510] core input error: VLC is unable to open the MRL 'file:///home/pi/p'.

shivasiddharth, 2017-11-07:
Can you cross check the radio station link that you have mentioned? VLC is trying to access "file:///home/pi/p" as radio station, which does not exist..

valascus, 2017-11-07:
This is what i have at actions.py

#Number of station names and station links should be the same
stnname=('Mega Hits')#Add more stations if you want
stnlink=('http://playerservices.streamtheworld.com/pls/KKNDFMAAC.pls')

shivasiddharth, 2017-11-07:
Can you try to stream the station from terminal using:
`vlc http://playerservices.streamtheworld.com/pls/KKNDFMAAC.pls`

valascus, 2017-11-07:
"vlc http://playerservices.streamtheworld.com/pls/KKNDFMAAC.pl                                                                                        s
VLC media player 2.2.6 Umbrella (revision 2.2.6-0-g1aae78981c)
[008f2830] pulse audio output error: PulseAudio server connection failure: Conne                                                                                        ction refused
[0090ae20] core interface error: no suitable interface module
[00858930] core libvlc error: interface "globalhotkeys,none" initialization fail                                                                                        ed
[0090ae20] dbus interface error: Failed to connect to the D-Bus session daemon:                                                                                         Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0090ae20] core interface error: no suitable interface module
[00858930] core libvlc error: interface "dbus,none" initialization failed
[00858930] core libvlc: Running vlc with the default interface. Use 'cvlc' to us                                                                                        e vlc without interface.
[008e7078] qt4 interface error: Could not connect to X server
[008e7078] skins2 interface error: cannot initialize OSFactory
[008e7078] [cli] lua interface: Listening on host "*console".
VLC media player 2.2.6 Umbrella
Command Line Interface initialized. Type `help' for help.
> [00868158] core playlist: stopping playback
[719009e0] http access: Raw-audio server found, m4a demuxer selected
[7190fb30] packetizer_mpeg4audio demux packetizer: AAC channels: 1 samplerate: 2                                                                                        2050
"

But it plays.

shivasiddharth, 2017-11-07:
I would really suggest you to use the GUI version. But here is what you can do.
In the actions.py file, after
`station=stnlink[num]`
add
`station='vlc '+ station`
and change
`p = subprocess.Popen(["/usr/bin/vlc",station],stdin=subprocess.PIPE,stdout=subprocess.PIPE)`
to
`os.system(station)`

valascus, 2017-11-07:
Still no sucess.

`
Nov 07 18:16:36 raspberrypi python[16405]: [71a00510] core input error: open of `file:///home/pi/p' failed
Nov 07 18:16:36 raspberrypi python[16405]: [71a00510] core input error: Your input can't be opened
Nov 07 18:16:36 raspberrypi python[16405]: [71a00510] core input error: VLC is unable to open the MRL 'file:///home/pi/p'. Check the log for details.
Nov 07 18:16:36 raspberrypi python[16405]: [00c9a108] [cli] lua interface: Listening on host "*console".
Nov 07 18:16:36 raspberrypi python[16405]: VLC media player 2.2.6 Umbrella
Nov 07 18:16:36 raspberrypi python[16405]: Command Line Interface initialized. Type `help' for help.
Nov 07 18:16:36 raspberrypi python[16405]: > Shutting down.
Nov 07 18:16:36 raspberrypi python[16405]: [00c9a108] [cli] lua interface: Requested shutdown.
`

valascus, 2017-11-07:
I just use the lite version becouse it makes no sense on having it, since terminal based is lighter and i don't really need a GUI to have a headless google asssitant

shivasiddharth, 2017-11-07:
Can you try playing it with mpv ? Try mpv from the command line and if it works, substitute vlc with mpv in actions.py. This is the best that i can do.

valascus, 2017-11-07:
Ok. I can play it like this: mpv --playlist=KKNDFMAAC.pls

i had to download the playlist first. not a big deal. now, what changes should i do to get this right?

valascus, 2017-11-07:
I've changed to station='mpv --playlist='+ station 

and the link is now
stnlink=('/home/pi/KKNDFMAAC.pls')

raspberrypi python[16952]: [playlist] Could not read directory.

Because it doens't get the path right.

├─16992 sh -c mpv --playlist=/*MISSING PATH
└─16993 mpv --playlist=/*MISSING PATH

shivasiddharth, 2017-11-07:
So am closing the issue..

valascus, 2017-11-07:
I'm still having the issue. Can't get the right path at mpv.
And it's still an issue not playing in the lite.

shivasiddharth, 2017-11-07:
did you try removing the /home/pi/ from the stnlink...

shivasiddharth, 2017-11-07:
You were saying that running headless version of desktop image did not make sense. Its been almost a day since you opened the two issues. Imagine the amount of time that you could have saved.. Now at least does it make sense?

valascus, 2017-11-07:
You're right. I could have went for the desktop version, but i got the minimal (only terminal).
They wouldn't have ocurred if i had installed the Raspbian desktop version. I've seen that you updated README but, for me, it doesn't make sense to have the desktop version, at least for this.
Could you help me solving this issue in the Lite version?

shivasiddharth, 2017-11-07:
I did my best but unfortunately, i don't think i can provide support on this issue any more and i am closing this issue.

closed 2017-11-07 by shivasiddharth