shivasiddharth/GassistPi #58: Youtube playback is not working.

connerwulf, 2017-11-10:
I did all of the steps in your read me file and everything worked flawless, except when I try and say "Play i got you" nothing happens. i waited to see it was the internet and it wasnt. The demo function showed the phrase was read but a no response was triggered.

shivasiddharth, 2017-11-10:
May i know as to which version of Raspbian you are using?

SUMemeLord, 2017-11-11:
I'm having the same issue. When I say "play" and then anything, it freezes for a moment and then restarts (plays a jingle and says "hi"). Everything else works fine except that.

This is what `cat /etc/os-release` shows:
`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

shivasiddharth, 2017-11-11:
Is the Raspbian Lite version or Desktop ?

SUMemeLord, 2017-11-11:
It's Desktop stretch

shivasiddharth, 2017-11-11:
Can you stop the service using:
`sudo systemctl stop gassistpi-ok-google.service`
now open a terminal and execute:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py`

Try the youtube command and let me know as to what happens. The youtube audio should play in the chromium browser.

SUMemeLord, 2017-11-11:
`pi@raspi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'Play Space Oddity'}
Playing: space oddity
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 122, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 111, in main
    YouTube(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 78, in YouTube
    playshell.stdin.write(bytes('/' + track + '\n1\n'))
TypeError: string argument without an encoding
pi@raspi:~ $ Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
BrokenPipeError: [Errno 32] Broken pipe
`

No browser opens on the Pi

shivasiddharth, 2017-11-11:
Please open the actions.py file and scroll down untill the youtube function, under the youtube function, you will have a line,  `playshell.stdin.write(bytes('/' + track + '\n1\n'))` replace it with `playshell.stdin.write(bytes('/' + track + '\n1\n','utf-8'))`

save and exit. Now if you start the google assistant youtube should play fine.

SUMemeLord, 2017-11-11:
Well, it does play and it works fine, but once it starts playing it won't stop, I can't say anything because the music is too loud, is there a way that the button that comes with the AIY kit can stop the playback?

shivasiddharth, 2017-11-11:
I shall create a service to stop radio/youtube music using that push-button. But in meanwhile, try this as a stop gap arrangement. Open a terminal and type:
`mpsyt set player vlc, set playerargs ,exit`
this will change the default player to vlc. now in the terminal type:
`vlc`
This will open up the vlc player. reduce the volume in the player say to something like 50-60 and close the player and restart assistant. next time, when you play the music, it will play at the set vlc volume.

SUMemeLord, 2017-11-11:
That works perfectly! Thank you for your time for both helping me and making the program, I greatly appreciate it!

shivasiddharth, 2017-11-11:
@SUMemeLord , I have created the stopbutton script for AIY users. Download [this file](https://github.com/shivasiddharth/GassistPi/blob/shivasiddharth-patch-2/src/stopbutton.py)  and place it in the src folder. Download [this file ](https://github.com/shivasiddharth/GassistPi/blob/shivasiddharth-patch-2/systemd/stopbutton.service)and place it in systemd folder in GassistPi. Execute the following commands one after the another.
```
sudo /home/pi/GassistPi/scripts/service-installer.sh    
sudo systemctl enable stopbutton.service  
sudo systemctl start stopbutton.service 
```
now you should be able to stop the music with that button. Kindly let me know if it works.

SUMemeLord, 2017-11-11:
Works perfectly! Thank you!

connerwulf, 2017-11-11:
What about using the button through GPIO? I’m not Using AIY

shivasiddharth, 2017-11-11:
@connerwulf It will still work. I have updated the git and README. The pushbutton has to be connected between GPIO23 and GRND.. I guess the issues have been sorted out. So i am closing this issue..

closed 2017-11-11 by shivasiddharth

connerwulf, 2017-11-13:
my is still not working. 
When I ran 

sudo systemctl stop gassistpi-ok-google.service
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py

it worked, so I added the functionality for the stop button and then enacted the headless startup script
and it did the same thing. after "hey google play i got you" its silent, so i ran the commands above again and added the "utf-8" line and tried the play command and the terminal outputed this.

home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py
ON_MUTED_CHANGED:
  {u'is_muted': False}
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {u'text': u'play I got you'}
Playing: i got you
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 113, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 102, in main
    YouTube(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 78, in YouTube
    playshell.stdin.write(bytes('/' + track + '\n1\n','utf-8'))
TypeError: str() takes at most 1 argument (2 given)
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
BrokenPipeError: [Errno 32] Broken pipe

shivasiddharth, 2017-11-13:
1. You are on python2 which does not require UTF-8. 
2. Make sure that you have connected the pushbutton properly. It has to be connected between gpio23 and grnd and also please make sure that the pushbutton terminals are not in continuous contact.

If you wish, you can format the SD card and try the installation again as I have upgraded the project to python3. I have also included the stop button.