shivasiddharth/GassistPi #59: File "/home/pi/GassistPi/src/main.py", line 39

n8thegr8, 2017-11-10:
I have one of the newer kits.  I am guessing that is why it's not working.   Maybe the pins are different.   Can you help?
Here is the error.
```
File "/home/pi/GassistPi/src/main.py", line 39
GPIO.setup(05,GPIO.OUT)
^
SyntaxError: invalid token
gassistpi-ok-google.service: main process exited, code=exited, status=1/FAILURE
Unit gassistpi-ok-google.service entered failed state.
gassistpi-ok-google.service holdoff time over, scheduling restart.
```

shivasiddharth, 2017-11-10:
It was mistake, i had failed to declare a gpio pin when i updated the project. Please open main.py using:
`sudo nano /home/pi/GassistPi/src/main.py`
add the line:
`GPIO.setup(25,GPIO.OUT)`
after the line that says "#Indicator Pins"

n8thegr8, 2017-11-13:
Awesome.   I got it working.   Thanks so much!

closed 2017-11-13 by shivasiddharth