shivasiddharth/GassistPi #62: Illegal instruction

ctatos, 2017-11-11:
Hi, I'm trying to install GassistPi on my raspberry pi zero w. I followed your README instructions to the letter but I keep hitting a "Illegal instruction" error when launching googlesamples-assistant-pushtotalk . Do you have an idea why?

shivasiddharth, 2017-11-11:
Looks like Google assistant was not fully installed. Can you run the Pi zero installer again. For pi zero w you need to run only the snowboy-deps-installer.sh and gassist-installer-pi-zero.sh. if you run into the same error, If you can copy paste the exact error it will be helpful.

ctatos, 2017-11-11:
Thanks for the quick reply shivasiddharth. The snowboy-deps-installer.sh went without any errors. However, I noticed 2 things with the gassist-installer-pi-zero.sh script:

Prior to mps-youtube: 
**mpris interface couldn't be initialized. Is dbus properly configured?**

and, more importantly, during the pyaudio installation:
Collecting pyaudio
  Using cached PyAudio-0.2.11.tar.gz
Building wheels for collected packages: pyaudio
  Running setup.py bdist_wheel for pyaudio ... error
  Complete output from command /home/pi/env/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-slzjghsv/pyaudio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp8jy9ngjspip-wheel- --python-tag cp35:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  **Failed building wheel for pyaudio**
  Running setup.py clean for pyaudio
**Failed to build pyaudio**
Installing collected packages: pyaudio
  Running setup.py install for pyaudio ... done
Successfully installed pyaudio-0.2.11

I can attach the full output if necessary...
Charles

shivasiddharth, 2017-11-11:
I am aware of these. They are just warnings none of them are threatening. If you are still getting error with the pushtotalk kindly post that errors ! Else you may close the issue if it was resolved.

ctatos, 2017-11-11:
Ah ok. Thanks shivasiddharth. Unfortunately I'm still hitting the error
"illegal instruction" once both script were excuted. Is there a way to
learn more on the cause of it (verbose?)

I also tried to install the Google Assistant gRPC API and I get the same
error if I execute the pushtotalk.py script...

Thanks for your help on this, I'm a bit of a newbie...

shivasiddharth, 2017-11-11:
Did the authentication process go through ? And are the pushtotalk commands run within environment. And as i said before, if you could copy paste the entire error only then it will be helpful.

ctatos, 2017-11-11:
Thanks again for your patience shivasiddharth...  The authentication process went through... and the pushtotalk command were ran from the env. 

I wish I could provide you with more information, but the command line simply shows:
(env) pi@raspberrypi:~/Project $ python -m pushtotalk 
Illegal instruction
(env) pi@raspberrypi:~/Project $ 

To be honest I am not sure where the problem is; I do not mind closing the thread... Have you recently ran the gRPC API on a Zero W? https://github.com/googlesamples/assistant-sdk-python/tree/master/google-assistant-sdk/googlesamples/assistant/grpc

shivasiddharth, 2017-11-11:
The command that you are using to run the pushtotalk is kind of off... 
Just open a new terminal and type the following commands one after the another.
```
source env/bin/activate 
googlesamples-assistant-pushtotalk  
```
These are already there in the README.
You will for sure have the pushtotalk working. This project uses gRPC API for Pi Zero and custom wakeword service and uses the assistant library for Pi 3 and Pi 2

ctatos, 2017-11-12:
Wow that worked! I'm baffled. Anyway, thanks shivasiddharth. Great support.

closed 2017-11-12 by ctatos