shivasiddharth/GassistPi #64: Exception when installing gassist-installer-pi3.sh

tpanicos, 2017-11-12:
Hi,

When running "sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh" i receive an exception.
Please see attached file - [Exception.txt](https://github.com/shivasiddharth/GassistPi/files/1464477/Exception.txt)

I am running this setup on a Raspberry Pi 2, so please let me know if I am running the wrong installer.

Thank you

tpanicos, 2017-11-12:
It looks like it may be an issue with the OAuth URL that the installer is calling? I'm no expert :)

shivasiddharth, 2017-11-12:
This is caused by temporary network snags. Please re-run the script.

tpanicos, 2017-11-12:
Yes you were right. Thank you!
One more question, where are the logs for this project kept?

shivasiddharth, 2017-11-12:
The logs of the google assistant activity are generally stored in the google's cloud console. If your issue is resolved, kindly close this thread.

closed 2017-11-13 by tpanicos