shivasiddharth/GassistPi #65: Import Errors in main.py and snowboy.py

patricktelling, 2017-11-12:
I am using an Raspberry Pi 3 B with Voice Hat, and the Raspbian os from google's AIY image

The main.py worked flawlesly without errors yestersday, but now I get this error while running main.py:

> pi@raspberrypi:~ $ python /home/pi/GassistPi/src/main.py

```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 25, in <module>
    import google.oauth2.credentials
ImportError: No module named google.oauth2.credentials
```

/////////////////////////////////////////////////////////////////////////////////////////////

In addition to main.py not working, I also could never get the snowboy service to start when running snowboy.py:

> pi@raspberrypi:~ $ python /home/pi/GassistPi/src/snowboy.py
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/snowboy.py", line 1, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 21, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 20, in swig_import_helper
    return importlib.import_module('_snowboydetect')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: libpython3.5m.so.1.0: cannot open shared object file: No such file or directory
```

shivasiddharth, 2017-11-12:
1. In the AIY image that google gives, the env i guess will be within the project folder. In my project, they need to be in the /home/pi/ directory.
2. The google assistant that you are using is running on Python2 i guess and the Python wrappers that i have issued are for Python3 that is what seems to be causing the snowboy issue. 

AIY image that google gives you have some minor tweakings. I would really not suggest the mixing the projects. This project works best when used with the default Raspbian Desktop image.

patricktelling, 2017-11-12:
Running a fresh non-AIY install worked, but now snowboy seems to be starting over and over again

shivasiddharth, 2017-11-12:
Stop the snowboy service, run the snowboy.py manually and paste the error here.

patricktelling, 2017-11-12:
> pi@raspberrypi-VoiceAI:~ $ python /home/pi/GassistPi/src/snowboy.py
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/snowboy.py", line 1, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 21, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 20, in swig_import_helper
    return importlib.import_module('_snowboydetect')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: dynamic module does not define init function (init_snowboydetect)
```

patricktelling, 2017-11-12:
It worked the first time, but entered a loop after I restarted the Pi

shivasiddharth, 2017-11-12:
Snowboy should be run within the environemnt. Use this cmd to run it manually:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/snowboy.py`

patricktelling, 2017-11-12:
Wow your're good, the issue was I had jarvis.umdl instead of jarvis.pmdl

Now its working! :D

patricktelling, 2017-11-12:
Everything seems good to go

patricktelling, 2017-11-12:
Oh, by the way, is there anyway to change the google assistant voice to male?

I am trying to make an Ironman like AI voice. (Jarvis)

shivasiddharth, 2017-11-12:
Its not possible at the moment. third party programs have to be used to change the pitch. As google's official update is long overdue, i dont want to spend days doing something which would be released as a feature in their next update (which is likely to have the choice of voice change). If your issue is resolved kindly close this thread.

closed 2017-11-12 by patricktelling