shivasiddharth/GassistPi #66: Illegal instruction

George-L, 2017-11-14:
Hello. I have a raspberry pi b+ and after the instalation, I try to run googlesamples-assistant-pushtotalk  comand and it says illegal instruction. and yes i was i the env mode. source env/bin/activate comand. What can be the problem?

shivasiddharth, 2017-11-14:
if you could copy paste the terminal contents only then it will give me a fair idea. Did the authentication go through ?

George-L, 2017-11-14:
Yes the authentication worked and now I formated the pi and i will try to install again. i the same error happens i will tell you.

George-L, 2017-11-15:
Please visit this URL to authorize this application: https://accounts.google.com                                                                             /o/oauth2/auth?response_type=code&client_id=1086385570949-l9jv64n8jgda4ir2j2nq6v                                                                             fausufp17t.apps.googleusercontent.com&redirect_uri=urn%3Aietf%3Awg%3Aoauth%3A2.0                                                                             %3Aoob&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&s                                                                             tate=CefhPMXFW8KeeXX6sT7SrUg0IbI0Pw&prompt=consent&access_type=offline
Enter the authorization code: 4/kkaxAwdE5eeHNeD8H-hRJ4f8vBa4MKr4vlLsujiSjNA
credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ googlesamples-assistant-pushtotalk
Illegal instruction
(env) pi@raspberrypi:~ $

I still have the same problem

shivasiddharth, 2017-11-15:
Looks like some Armv7 packages have been installed. May I know as to which installer you used.?

George-L, 2017-11-15:
I used this 2 comands to install.

sudo  /home/pi/GassistPi/scripts/snowboy-deps-installer.sh
sudo  /home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh

shivasiddharth, 2017-11-15:
Within the environment, try running this command: 
python -m pip install --upgrade google-assistant-sdk[samples]

And then try the pushtotalk

dswillows, 2017-11-18:
Hello
same issue, pi zero w, new install, all steps good until same point. Tried last suggestion, same result...

shivasiddharth, 2017-11-19:
Can you check if you have the pushtotalk.py file in the assistant library ? It should be in the environment under /googlesamples/assistant/grpc folder.

dswillows, 2017-11-19:
Hello, this is true for me;

(env) pi@LEWA:~/env/lib/python3.5/site-packages/googlesamples/assistant/grpc $ ls
assistant_helpers.py  audio_helpers.py  __init__.py  pushtotalk.py  __pycache__

shivasiddharth, 2017-11-19:
As long as you have this pushtotalk.py file, you can run the snowboy. It should not be an issue. Are you using the AIY image by any chance ?

dswillows, 2017-11-19:
snowboy? this is step 4.2.7 in guide (test google assistant)
No to AIY, using USB-DAC

shivasiddharth, 2017-11-19:
I am asking you to proceed further. The assistant will work. 4.2.7 is not compulsory.

dswillows, 2017-11-19:
ok, so proceed to next step, and enabled as a service. 
Alas, no chime on reboot as I've had in previous installs... not sure what info is useful to diagnose...

shivasiddharth, 2017-11-19:
Stop the service using:
`sudo systemctl stop snowboy.service   `
Run the following command manually and post the output:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/snowboy.py`

dswillows, 2017-11-19:
Ok
/home/pi/env/bin/python -u /home/pi/GassistPi/src/snowboy.py
Illegal instruction

shivasiddharth, 2017-11-19:
Ok, thanks, i will take a look as to what is causing this issue.

dswillows, 2017-11-19:
Thank you sir

shivasiddharth, 2017-11-19:
One final question ! Can you tell me as to what OS you are using ?

dswillows, 2017-11-19:
2017-09-07-raspbian-stretch
Linux 4.9.59+ #1047 Sun Oct 29 11:47:10 GMT 2017 armv6l GNU/Linux
zero w

chrisarcher, 2017-11-25:
Hi,
I have this error too, on a Zero W with latest stretch Raspbian and ReSpeaker 2-mic HAT.

shivasiddharth, 2017-11-26:
Guys, I have fixed the bug. Please try again using a new copy of the OS. I have also changed the order of installation. First use the pi zero installer and then finally the snowboy dependencies.

closed 2017-11-26 by shivasiddharth