shivasiddharth/GassistPi #67: GassistPi working with systemctl start but not on bootup

KGunther, 2017-11-15:
Hi,
   I have installed GassistPi following your instructions (not using snowboy). I have enabled the service but it does not work (systemctl list-units shows it as loaded,failed). If I start it from an SSH session using "sudo systemctl start gassistpi-ok-google.service" it works fine.

shivasiddharth, 2017-11-15:
Are you using the Desktop version of Raspbian or the CLI Lite version? Please run the systemctl status command of the assistant and post the results here.

KGunther, 2017-11-16:
Here is the result of systemctl status
Last login: Thu Nov 16 00:34:18 2017 from 172.20.3.64
pi@KWCpi:~ $ sudo systemctl status
● KWCpi
    State: degraded
     Jobs: 0 queued
   Failed: 1 units
    Since: Wed 1969-12-31 19:00:01 EST; 47 years 10 months ago
   CGroup: /
           ├─user.slice
           │ └─user-1000.slice
           │   ├─user@1000.service
           │   │ ├─gvfs-gphoto2-volume-monitor.service
           │   │ │ └─760 /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
           │   │ ├─dbus.service
           │   │ │ └─600 /usr/bin/dbus-daemon --session --address=systemd: --nof
           │   │ ├─gvfs-udisks2-volume-monitor.service
           │   │ │ └─745 /usr/lib/gvfs/gvfs-udisks2-volume-monitor
           │   │ ├─gvfs-mtp-volume-monitor.service
           │   │ │ └─765 /usr/lib/gvfs/gvfs-mtp-volume-monitor
           │   │ ├─gvfs-goa-volume-monitor.service
           │   │ │ └─774 /usr/lib/gvfs/gvfs-goa-volume-monitor
           │   │ ├─gvfs-afc-volume-monitor.service
           │   │ │ └─769 /usr/lib/gvfs/gvfs-afc-volume-monitor
           │   │ ├─init.scope
           │   │ │ ├─538 /lib/systemd/systemd --user
           │   │ │ └─542 (sd-pam)
           │   │ └─gvfs-daemon.service
           │   │   ├─650 /usr/lib/gvfs/gvfsd
           │   │   ├─660 /usr/lib/gvfs/gvfsd-fuse /run/user/1000/gvfs -f -o big_
           │   │   └─812 /usr/lib/gvfs/gvfsd-trash --spawner :1.4 /org/gtk/gvfs/
           │   ├─session-c1.scope
           │   │ ├─474 /bin/login -f
           │   │ └─549 -bash
           │   ├─session-c2.scope
           │   │ ├─579 lightdm --session-child 14 17
           │   │ ├─591 /usr/bin/lxsession -s LXDE-pi -e LXDE
           │   │ ├─644 /usr/bin/ssh-agent x-session-manager
           │   │ ├─668 openbox --config-file /home/pi/.config/openbox/lxde-pi-rc
           │   │ ├─676 lxpolkit
           │   │ ├─679 lxpanel --profile LXDE-pi
           │   │ ├─685 pcmanfm --desktop --profile LXDE-pi
           │   │ ├─693 /usr/bin/ssh-agent -s
           │   │ └─731 /usr/lib/menu-cache/menu-cached /run/user/1000/menu-cache
           │   └─session-c5.scope
           │     ├─1161 sshd: pi [priv]
           │     ├─1171 sshd: pi@pts/0
           │     ├─1174 -bash
           │     ├─1188 sudo systemctl status
           │     ├─1192 systemctl status
           │     └─1193 pager
           ├─init.scope
           │ └─1 /sbin/init splash
           └─system.slice
             ├─lightdm.service
             │ ├─469 /usr/sbin/lightdm
             │ └─482 /usr/lib/xorg/Xorg :0 -seat seat0 -auth /var/run/lightdm/ro
             ├─systemd-timesyncd.service
             │ └─308 /lib/systemd/systemd-timesyncd
             ├─vncserver-x11-serviced.service
             │ ├─448 /usr/bin/vncserver-x11-serviced -fg
             │ ├─466 /usr/bin/vncserver-x11-core -service
             │ ├─513 /usr/bin/vncagent service 14
             │ ├─716 /usr/bin/vncserverui service 15
             │ └─728 /usr/bin/vncserverui -statusicon 7
             ├─dbus.service
             │ └─355 /usr/bin/dbus-daemon --system --address=systemd: --nofork -
             ├─hciuart.service
             │ └─506 /usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - b8:
             ├─udisks2.service
             │ └─752 /usr/lib/udisks2/udisksd --no-debug
             ├─ssh.service
             │ └─452 /usr/sbin/sshd -D
             ├─avahi-daemon.service
             │ ├─342 avahi-daemon: running [KWCpi.local]
             │ └─384 avahi-daemon: chroot helper
             ├─triggerhappy.service
             │ └─345 /usr/sbin/thd --triggers /etc/triggerhappy/triggers.d/ --so
             ├─systemd-logind.service
             │ └─395 /lib/systemd/systemd-logind
             ├─polkit.service
             │ └─707 /usr/lib/policykit-1/polkitd --no-debug
             ├─cron.service
             │ └─348 /usr/sbin/cron -f
             ├─systemd-udevd.service
             │ └─152 /lib/systemd/systemd-udevd
             ├─rsyslog.service
             │ └─339 /usr/sbin/rsyslogd -n
             ├─bluetooth.service
             │ └─511 /usr/lib/bluetooth/bluetoothd
             ├─bluealsa.service
             │ └─514 /usr/bin/bluealsa
             ├─systemd-journald.service
             │ └─127 /lib/systemd/systemd-journald
             └─dhcpcd.service
               ├─378 /sbin/dhcpcd -q -b
               └─419 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf
pi@KWCpi:~ $

Here is the result of a systemctl list-units
pi@KWCpi:~ $ sudo systemctl list-units gas*
  UNIT                        LOAD   ACTIVE SUB    DESCRIPTION
● gassistpi-ok-google.service loaded failed failed Google Assistant

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

1 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
pi@KWCpi:~ $

shivasiddharth, 2017-11-16:
can you post the output of `systemctl start gassistpi-ok-google.service`

shivasiddharth, 2017-11-16:
have you set the pi to auto login on boot ?

KGunther, 2017-11-16:
Thank you for your support.
User pi does login on boot, when I boot the system an attached screen will show the GUI desktop for user pi

pi@KWCpi:~ $ systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-11-16 11:40:36 EST; 3h 5min ago
  Process: 866 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py (code=exited, status=1/FAILURE)
 Main PID: 866 (code=exited, status=1/FAILURE)

Nov 16 11:40:35 KWCpi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Nov 16 11:40:36 KWCpi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, scheduling restart.
Nov 16 11:40:36 KWCpi systemd[1]: Stopped Google Assistant.
Nov 16 11:40:36 KWCpi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Nov 16 11:40:36 KWCpi systemd[1]: Failed to start Google Assistant.
Nov 16 11:40:36 KWCpi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Nov 16 11:40:36 KWCpi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
pi@KWCpi:~ $

shivasiddharth, 2017-11-16:
Can you open a terminal, run the code manually:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py`
Hope you have not changed the directory of the project !

KGunther, 2017-11-16:
Thank you for your help. I have posted the information on Github.

     I do not claim to be a Python expert but I am trying to learn. 
Reading the code for main.py I see
     parser.add_argument('--credentials', type=existing_file,
                         metavar='OAUTH2_CREDENTIALS_FILE',
                         default=os.path.join(
                             os.path.expanduser('/home/pi/.config'),
                             'google-oauthlib-tool',
                             'credentials.json'
                         )
  does this mean that it is looking for 
/home/pi/.config/credentials.json? I have the google authentication file 
as /home/pi/assistant.json.

                                         Ken Gunther

shivasiddharth, 2017-11-16:
This credentials.json is the file which has details of your authentication code that you pasted during installation. Both are different.

KGunther, 2017-11-17:
When I go to the Pi by SSH and run the command "/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py" the assistant runs and logs each event to the terminal.

shivasiddharth, 2017-11-17:
I thought so. The services were starting and getting timed out even before the network started.

Open the gassistpi-ok-google.service file inside the /home/pi/GassistPi/systemd

After the line `Description=Google Assistant` remove the `After=network.target`  and ddd the following:
```
Requires=network-online.target
After=network.target network-online.target multi-user.target
```
Now, you can use the supplied main.py file.

Since this issue is fixed I am closing this thread.

closed 2017-11-17 by shivasiddharth