shivasiddharth/GassistPi #68: Works but not for custom wake word

compt1ci, 2017-11-15:
I ran through this setup from start to finish. I started with a fresh flash from the Google AIY site. I then used the git command to clone to your image here. Added a custom startup sound and added a custom.pmdl file to the folder with the rest of the wake words, but “Ok-Google” is the only wake word that works. Is this part of the known issue with Google? Any help would be great! Thanks in advance

compt1ci, 2017-11-15:
By the way, I added the custom wake word and then proceeded with the rest of the steps. I figured it may be important to make sure the .pmdl file is there from the start.

shivasiddharth, 2017-11-15:
Just adding the wakeword to the folder will not do, you need to change the wakeword in the script. Please open the snowboy.py file and scroll to a line that says models. change the replace the existing model names with your custom created model name. Path should not not be changed just the model name alone.

compt1ci, 2017-11-15:
Thank you! I found two options in that file, for alexa.umdl and snowboy.umdl. I replaced the alexa model name to custom.pmdl for the file that I've added. Also used a systemctl stop command for all 3 services, and started those again. Once again, it works for ok-google but not the custom command. 

Sorry if I'm missing something here, as I am kind of new to the environment. Thanks again!

shivasiddharth, 2017-11-15:
Please post the outputs of:
```
sudo nano /etc/asound.conf
sudo nano ~/.asoundrc
```
Stop the service and run the snowboy manually using:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/snowboy.py`
and post that output as well

compt1ci, 2017-11-15:
Included them all in the following txt file. Thanks!

[Update.txt](https://github.com/shivasiddharth/GassistPi/files/1476542/Update.txt)

shivasiddharth, 2017-11-15:
I dont remember in which thread i posted. But here is the issue, running this project on an AIY image is not advisable. It is causing dependency issues.  Please use the regular Raspbian Desktop image from the Pi website.

compt1ci, 2017-11-16:
Good call! Fresh flash of Raspbian did the trick! Thanks for all the help!

closed 2017-11-16 by compt1ci