shivasiddharth/GassistPi #70: I tried multiple times but failed. Ready to pay for a image file.

rarenu, 2017-11-15:
Dear Mr. Sid, I tried many times but didn't make it work. Whereas the original Voice Hat image works fine. Please help me with an image file and I will pay for it. I will enter my Google and Wifi credentials to use it. At my 70 I am rest less with out proper sleep struggling with it to see it is working. Thank you in advance.

shivasiddharth, 2017-11-15:
Can you please describe as to what exactly is the issue ? 
Please do not use the Voice HAT image with this project. In the AIY image, environment is in another directory and this project requires the environment to be in another directory.

rarenu, 2017-11-15:
Dear Mr Sid, at the stage of testing the sound the start menu goes off. I am not using voice hat setup. Using only yours as you instructed. I tried all the options suggested. ragoth@gmail.com. Please help me with an image file. As I said I will pay for it.

rarenu, 2017-11-15:
Dear mr. Sid, the start menu goes off at the stage of testing the sound
card while installation.
I tried all the options suggested by you but nothing works out. Please
share the image through Dropbox or Google drive. I will pay.
Thank you in advance.

closed 2017-11-15 by shivasiddharth