shivasiddharth/GassistPi #71: ImportError: libpython3.5m.so.1.0: cannot open shared object file: No such file or directory

dudukk1312, 2017-11-18:
Dear shivasiddharth,

Do you remember me. I tried to format SD card and re-install everything of this project. You said it support python3 now. But when i try to run snowboy.py. This error raised.
```
(env) pi@raspberrypi:~/GassistPi/src $ sudo /home/pi/env/bin/python -u /home/pi/GassistPi/src/snowboy.py
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/snowboydetect.py", line 18, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1191, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1161, in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 539, in _check_name_wrapper
  File "<frozen importlib._bootstrap>", line 1715, in load_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
ImportError: libpython3.5m.so.1.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/snowboy.py", line 1, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/pi/GassistPi/src/snowboydetect.py", line 21, in <module>
    _snowboydetect = swig_import_helper()
  File "/home/pi/GassistPi/src/snowboydetect.py", line 20, in swig_import_helper
    return importlib.import_module('_snowboydetect')
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libpython3.5m.so.1.0: cannot open shared object file: No such file or directory
```
I didn't modify anything, just follow your tutorial in Github page. So the snowboy service was failed to start.

shivasiddharth, 2017-11-18:
For any updation on the lines of python versions, you need to re-install google assistant. This error is either because of environment being out of place or because of assistant running python2. 
Are you using the AIY image?

dudukk1312, 2017-11-18:
Yes, I'm using it. When check python version, it's 3.4.2. 
I format and re-install clearly AIY image for it, then update lastest everthing. Next, I followed step by step your guideline in this page. Every thing was successfully. I could run sample pustotalk smoothly, but when run snowboy.py. This error make it failed.

shivasiddharth, 2017-11-18:
This issue has already been addressed. Please check this #65

closed 2017-11-18 by shivasiddharth