shivasiddharth/GassistPi #73: Custom Wake Word working only once  - Ubuntu Installation

bimalnair, 2017-11-25:
Hi, I have earlier installed GA as per your instructions in RPI and all running fine. I wanted to try the same in Ubuntu 16.04  as I wanted GA with custom actions for my another project. I have managed to install and customized the scripts  by commenting out the GPIO parts. The gassistpi-ok-google service is running fine and the Youtube and Radio custom actions are working without any problem. But when I wanted to run Snowboy service I ended up getting errors and now I have made the Python wrapper with the Python3-Config and SWIG 3.0.12 and generated the Snowboy files. Now when I run I don't get any errors but custom wake word works only once, I need to restart again to make it work one more time. 

Can you please let me know if I need to make any other changes , to make it work continuously. If I check the status its always active and running.

● snowboy.service - Google Assistant
   Loaded: loaded (/lib/systemd/system/snowboy.service; enabled; vendor preset: 
   Active: active (running) 
 Main PID: 2160 (python)
   CGroup: /system.slice/snowboy.service
           └─2160 /home/env/bin/python -u /home/bimal/GassistPi/src/snowbo

systemd[1]: Started Google Assistant.
python[2160]: Listening... Press Ctrl+C to exit
python[2160]: INFO:snowboy:Keyword 1 detected at time: 201
python[2160]: INFO:assistant:Recording audio request.
python[2160]: INFO:assistant:End of audio request detected
python[2160]: INFO:assistant:Finished playing assistant re

shivasiddharth, 2017-11-25:
If you look into the snowboy.py file you can see that I am stopping the detector on wakeword detection and restarting it again after GA interaction. This stopping and starting works only in python3 on raspbian and not in Ubuntu on pi.
If you want continuous listening, you can remove the terminate command under detected function. But it will cause GA to function erratically and make the custom wakeword service susceptible to crashes.

closed 2017-11-25 by shivasiddharth

bimalnair, 2017-11-27:
Sid, Thanks for your reply, I am using this on Ubuntu 16.04 in my laptop and not in pi, is there any workaround to make wakeword detection restart  after GA interaction in this environment. As the detection is flawless first time so if we are able to restart it should work without any problem. Any guidance in this regard will be a big help.

shivasiddharth, 2017-11-27:
replace the detector.start under the detected function to break. The systemd will restart a service if it crashes. So basically you have the snowboy service restarting after every interaction.