shivasiddharth/GassistPi #74: Home Assistant Compatible?

helloworld2k, 2017-11-29:
Hi Shiva,

Thanks for upgrading the project to the Python 3.
I currently use Home Assistant which is open source automation platform on Raspberry Pi, Google Assistant is one of the add-on, may I know is there any way I can put the GassistPi to this platform? I quite like your project rather than the original, thank you.

Regards,
Jun

shivasiddharth, 2017-12-03:
You need to modify the "pi" user accounts mentioned in this code to make it work in Home assistant. In the next update, i will try and include home assistant with the google assistant.

closed 2017-12-03 by shivasiddharth

Electronlibre2012, 2021-04-02:
> Hi Shiva,
>
> Thanks for upgrading the project to the Python 3.
> I currently use Home Assistant which is open source automation platform on Raspberry Pi, Google Assistant is one of the add-on, may I [...]

Hello @helloworld2k ,

very interested to add the Gassistpi to a media_player entity in HA!! Have you succeed in to see it in HA please?

Thanks