shivasiddharth/GassistPi #78: Help required

Baidya99, 2017-12-03:
I am using a usb sound card and analog mic. Even though I can use it without any driver on windows I am unable to use it on the pi.
Raspberry pi is is detecting the sound card but when testing the mic it it not being able to take input.
Could you please help.

shivasiddharth, 2017-12-03:
Did you install the USB-DAC alsa configuration that i have given in the git ?

Baidya99, 2017-12-03:
Yes I used that config

shivasiddharth, 2017-12-03:
post the output of asoundrc and asound.conf !

Baidya99, 2017-12-03:
asoundrc:

`pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
`

Baidya99, 2017-12-03:
**asoundrc:**
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

Baidya99, 2017-12-03:
**asound.conf:**
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

Baidya99, 2017-12-03:
The speaker connected to usb sound card is working but the mic does not

shivasiddharth, 2017-12-03:
Check your mic volume in alsamixer and also try removing `channels 1` from both asoundrc and asound.conf

Baidya99, 2017-12-03:
Okay

Baidya99, 2017-12-03:
Still not working.

shivasiddharth, 2017-12-03:
Try recording an audio using arecord 
`arecord --device=hw:1,0 --format S16_LE --rate 44100 -c1 test.wav`
stop the recording  manually and playback the recorded audio.

Baidya99, 2017-12-03:
how to playback?

shivasiddharth, 2017-12-03:
aplay test.wav

Baidya99, 2017-12-03:
okay

Baidya99, 2017-12-03:
Getting error:
arecord: main:788: audio open error: Device or resource busy

shivasiddharth, 2017-12-03:
I think the google assistant service is running which is making use of the mic. What do you mean by "Raspberry pi is is detecting the sound card but when testing the mic it it not being able to take input"? What exactly is your issue ?

Baidya99, 2017-12-03:
The assistant service is not running because I have still not executed the command to make it autostart on boot and I had rebooted after carrying out the steps in Continuing after setting up audio.

shivasiddharth, 2017-12-03:
then what do you mean by "when testing the mic it it not being able to take input"? how did you test it ?

Baidya99, 2017-12-03:
By going to [https://www.onlinemictest.com/](https://www.onlinemictest.com/)
I did it before cloning your repository.

shivasiddharth, 2017-12-03:
Before cloning my repository, you would not have had the audio config installed. USB-DACs in pi are not plug an play. Meaning, you need the alsa configs for mic to work. In pi and linux, you test mics with arecord. Use this command to record your mic and i am pretty much sure that it will work.
`arecord -d 10 -D hw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`
you need not stop the recording manually, it will automatically record samples only for 10 seconds. play the recorded samples using:
`aplay /home/pi/mic-test.wav`

Baidya99, 2017-12-03:
Nope still the same error

shivasiddharth, 2017-12-03:
Try this arecord with plugin !
`arecord -d 10 -D plughw:1,0 -r 16000 -f S16_LE -c1 /home/pi/mic-test.wav`

Baidya99, 2017-12-03:
It is recording now but only noise is being recorded.

shivasiddharth, 2017-12-03:
You need to look into your mic. I am closing this issue now.

closed 2017-12-03 by shivasiddharth