shivasiddharth/GassistPi #80: Youtube streaming with voice command

Tony1957, 2017-12-04:
Hi Sid, you do great work and should be commended for it. 
I am not a coder but i am good at following instructions. 
I put your latest version of GassistPi on both my raspberry pi's 2 and 3 and works great. a little bit better on the pi3. 
I have a question concerning the you-tube streams. they are working great on both pi's for music playing, But my idea is to mount a monitor on my kitchen wall with built in pi3 and small amp for the sound. 
Is it possible to stream you-tube videos to that monitor by voice command, the reason is my wife likes cooking and it's to stream cooking recipes by voice command with the ability to pause by voice as she follows along.

mylife4aiurr, 2017-12-12:
Similar question. I have a google chrome cast attached to a TV in my bedroom. In the google assistant app on my phone the chrome is named "parents TV". So I would like to be able to issue a command something like:
AIY pi
"OK Google, play cat videos on the parents TV"
AIY cant do it.
"OK Google, trigger shutdown" <<<<--- AIY pi now off

On my phone:
"OK Google, play cat videos on the parents TV"
works like charm.

shivasiddharth, 2017-12-12:
@mylife4aiurr , I am not sure about the AIY cloudspeech or the associated branches, but last time when i checked for Youtube streaming to chromecast from Google Assistant on Pi (this project), it was working fine.

mylife4aiurr, 2017-12-12:
@shivasiddharth
Ok so maybe I'm issuing wrong voice command. What sentence structure would you use for aiy Google to cast video to Chromecast.. statement above didn't work for me.

shivasiddharth, 2017-12-12:
@mylife4aiurr , in your previous post you just mentioned "AIY cant do it" it did not give me enough info on the issue. This edited version tells me as to what is happening. In this project i have associated the "play" command to play audio from youtube videos locally on pi.  You have two options:

1. Change the trigger word for local Youtube music to something other than "play", for example stream. Open main.py file and scroll to a line that says  `if 'play'.lower() in str(usrcmd).lower():` change play to something else, save and close.
or
2. Instead of saying "play cat videos on the parents TV" you can say "youtube cat videos on the parents TV"

mylife4aiurr, 2017-12-12:
I'm not home at the moment to change the main.py file. But I called my wife to try your 2nd suggestion. 
She said:
 "Ok Google, YouTube cat videos on parents tv"
Response:
"Sorry I'm not sure how to help"

shivasiddharth, 2017-12-13:
@mylife4aiurr , I just checked my chromecast. The streaming does not seem to work now.

mylife4aiurr, 2017-12-13:
Thanks for checking. I'd feel like I wasted your time if it worked for you. 
In the Google assistant app u can link the following video services:
Netflix
HBO Now
CBS
Viki
And of course YouTube
I've already linked my Netflix account, so would be great to say something like "ok Google stream Netflix breaking bad on parents tv" in addition to YouTube videos

shivasiddharth, 2017-12-13:
@mylife4aiurr I am half way through Kodi integration, will look into these once thats done.

shivasiddharth, 2017-12-19:
@Tony1957 , I have enabled youtube streaming using voice requests. In the actions file, you can choose to either play the video on kodi or locally on pi. Ref:  https://youtu.be/ahc6XKYriHo

Tony1957, 2017-12-19:
Hi Sid
I have an issue, i installed the recent gassistpi with the kodi on it,  installation was fine no issue's
when i entered, source env/bin/activate - google-assistant-demo, i asked a few commands and she answered ok.

I carried on with the headless codes, restarted pi and i heared the starting audio and my mic light on usb
was flashing as it normally does, when i issue a command like ( what time is it ) she tries to answer, but the light stops flashing and stays on, on the mic and the starting audio plays, it goes in the same loop on all commands.

I tried a fresh install with the latest raspbian and it is still the same, it is a pi 2 i am testing it on if it makes an difference.
any help would be great, thanks in advance

shivasiddharth, 2017-12-19:
This seems like a server related issue to me. A slow internet or segmentation fault kind of a thing. If you have not entered the YouTube API Key value, that could also be a reason. Stop the services using: 
`sudo systemctl stop gassistpi-ok-google.service
`
 And try manually running the main.py. Whatever is causing the assistant to crash, the reason can be seen on the screen.

Tony1957, 2017-12-19:
Hi Sid

Thank for your quick response
i got the api key in it's place
i ran main.py and got

pi@raspberrypi:~ $ /home/pi/GassistPi/src/main.py
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 17, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ImportError: No module named kodijson
pi@raspberrypi:~ $

Regards
Tony

shivasiddharth, 2017-12-19:
Kodi json should have got installed, when the installer script was run. Anyways, move into the env using:
`source env/bin/activate`
and now execute:
`sudo pip install kodi-json` 

After that, try running the main.py

shivasiddharth, 2017-12-19:
The kodi-json would be installed for root users. So when you run the main.py, make sure to add a sudo

Tony1957, 2017-12-19:
Hi Sid

I tried as you said, sudo did not work for me, i had to change permissions
on the main.py file
and i got,

pi@raspberrypi:~ $ sudo /home/pi/GassistPi/src/main.py
sudo: /home/pi/GassistPi/src/main.py: command not found
pi@raspberrypi:~ $ /home/pi/GassistPi/src/main.py
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 24, in <module>
    import google.oauth2.credentials
ImportError: No module named google.oauth2.credentials
pi@raspberrypi:~ $

Maybe i should try another fresh install

shivasiddharth, 2017-12-19:
The main.py should be run within the environment. Looks like you are running it outside environment. That is why it is not picking up any of those packages. Execute this line, the environment stuff is taken care of automatically.
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py`

One last question ? did you replace the main.py with kodi-exclusive main.py ? Because, the kodi-exclusive will not even work for general queries like time and weather, as the name suggests its only for Kodi just with an added advantage of you not having the need to say on kodi everytime. I just realized that and have added a note in Readme.

Tony1957, 2017-12-19:
Hi Sid

I replaced the files main.py and assistant.py
i ran the link you gave me, and got,

ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py",
line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File
"/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line
83, in create_connection
    raise err
  File
"/home/pi/env/lib/python3.5/site-packages/urllib3/util/connection.py", line
73, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File
"/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line
601, in urlopen
    chunked=chunked)
  File
"/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line
357, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.5/http/client.py", line 1107, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python3.5/http/client.py", line 1152, in _send_request
    self.endheaders(body)
  File "/usr/lib/python3.5/http/client.py", line 1103, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python3.5/http/client.py", line 934, in _send_output
    self.send(msg)
  File "/usr/lib/python3.5/http/client.py", line 877, in send
    self.connect()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py",
line 166, in connect
    conn = self._new_conn()
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/connection.py",
line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection
object at 0x74e7a270>: Failed to establish a new connection: [Errno 113] No
route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py",
line 440, in send
    timeout=timeout
  File
"/home/pi/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line
639, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.5/site-packages/urllib3/util/retry.py",
line 388, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.1.15',
port=8080): Max retries exceeded with url: /jsonrpc (Caused by
NewConnectionError('<urllib3.connection.HTTPConnection object at
0x74e7a270>: Failed to establish a new connection: [Errno 113] No route to
host',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 150, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 122, in main
    process_event(event)
  File "/home/pi/GassistPi/src/main.py", line 72, in process_event
    status=mutevolstatus()
  File "/home/pi/GassistPi/src/actions.py", line 211, in mutevolstatus
    status= kodi.Application.GetProperties({"properties":
("volume","muted")})
  File "/home/pi/env/lib/python3.5/site-packages/kodijson/kodijson.py",
line 121, in hook
    return self.kodi.execute(kodimethod, *args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/kodijson/kodijson.py",
line 71, in execute
    auth=(self.username, self.password))
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line
112, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/api.py", line 58,
in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py",
line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/sessions.py",
line 618, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/requests/adapters.py",
line 508, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError:
HTTPConnectionPool(host='192.168.1.15', port=8080): Max retries exceeded
with url: /jsonrpc (Caused by
NewConnectionError('<urllib3.connection.HTTPConnection object at
0x74e7a270>: Failed to establish a new connection: [Errno 113] No route to
host',))
pi@raspberrypi:~ $

shivasiddharth, 2017-12-19:
Ok, I just realized as to what was happening. I had programmes the Kodi volume to be muted on wakeword detection. If someone is not running kodi, then it will search of that kodi to mute the volume and as it does not find any, it will crash. Thanks a lot for reporting this.  I have now, commented the volume changing part. It should run fine now if kodi is not running. 
Can you remove the existing GassistPi folder and clone the git again  and try.

Tony1957, 2017-12-19:
Hi Sid

Ok i will do a fresh install now.

Regards
Tony

shivasiddharth, 2017-12-19:
@Tony1957 , Apologies for the mess..

Tony1957, 2017-12-19:
Hi Sid
I have done a fresh install, everything worked as it should, i could issue
commands and she would answer them.

I want to use vlc instead of kodi, so i have uncommented and commented the
lines.

When i replaced the main.py and assistant.py it started crashing again, i
ran the main.py again
and got,

pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ /home/pi/GassistPi/src/main.py
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connection.py",
line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File
"/home/pi/env/lib/python3.4/site-packages/urllib3/util/connection.py", line
83, in create_connection
    raise err
  File
"/home/pi/env/lib/python3.4/site-packages/urllib3/util/connection.py", line
73, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File
"/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line
601, in urlopen
    chunked=chunked)
  File
"/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line
357, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.4/http/client.py", line 1090, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python3.4/http/client.py", line 1128, in _send_request
    self.endheaders(body)
  File "/usr/lib/python3.4/http/client.py", line 1086, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python3.4/http/client.py", line 924, in _send_output
    self.send(msg)
  File "/usr/lib/python3.4/http/client.py", line 859, in send
    self.connect()
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connection.py",
line 166, in connect
    conn = self._new_conn()
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/connection.py",
line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection
object at 0x76098b10>: Failed to establish a new connection: [Errno 113] No
route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.4/site-packages/requests/adapters.py",
line 440, in send
    timeout=timeout
  File
"/home/pi/env/lib/python3.4/site-packages/urllib3/connectionpool.py", line
639, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/lib/python3.4/site-packages/urllib3/util/retry.py",
line 388, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.1.15',
port=8080): Max retries exceeded with url: /jsonrpc (Caused by
NewConnectionError('<urllib3.connection.HTTPConnection object at
0x76098b10>: Failed to establish a new connection: [Errno 113] No route to
host',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 150, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 122, in main
    process_event(event)
  File "/home/pi/GassistPi/src/main.py", line 72, in process_event
    status=mutevolstatus()
  File "/home/pi/GassistPi/src/actions.py", line 211, in mutevolstatus
    status= kodi.Application.GetProperties({"properties":
("volume","muted")})
  File "/home/pi/env/lib/python3.4/site-packages/kodijson/kodijson.py",
line 121, in hook
    return self.kodi.execute(kodimethod, *args, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/kodijson/kodijson.py",
line 71, in execute
    auth=(self.username, self.password))
  File "/home/pi/env/lib/python3.4/site-packages/requests/api.py", line
112, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/api.py", line 58,
in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py",
line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/sessions.py",
line 618, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/lib/python3.4/site-packages/requests/adapters.py",
line 508, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError:
HTTPConnectionPool(host='192.168.1.15', port=8080): Max retries exceeded
with url: /jsonrpc (Caused by
NewConnectionError('<urllib3.connection.HTTPConnection object at
0x76098b10>: Failed to establish a new connection: [Errno 113] No route to
host',))
(env) pi@raspberrypi:~ $

shivasiddharth, 2017-12-19:
The crash is because, you dont have an active Kodi running. So you want to just view the Youtube video on VLC ? without all that Kodi and stuff ? I have modified both the files for you. You can download them from here  https://github.com/shivasiddharth/GassistPi/tree/master/Extras/Youtube%20Video%20on%20VLC and replace the existing ones.
By using syntax: From Youtube play videoname 
you should now have the video playing in VLC. Hope this helps.

Tony1957, 2017-12-19:
> master/Extras/Youtube%20Video%20on%20VLC and replace the existing ones.

Hi Sid

I have replaced the files you made for me, now i got her talking again
thanks,
I have set my pi up on a monitor which has vlc  do i need to configure vlc
in any way.

I tried to get it to stream to vlc but failed so far, i tried the phrase
'play' but that didn't work
so i tried the prase 'stream' that works for the music, how do i get a
video to play, for 'eg' one of your videos,
not sure if i have to put a streaming link into vlc or not.

I had noticed when i open vlc manually to see if it plays anything, when
the music from youtube is playing and i say, Ok Google, stop it
automatically closes vlc.

Regards
Tony

shivasiddharth, 2017-12-19:
Command is: "From Youtube play" 
I felt this was more conversational. Right now, you cannot pause VLC by voice, as you found out, you can only stop it. In the next update, i will add google music, at that time, will add voice controls to vlc, like pause, next, previous etc.

Tony1957, 2017-12-19:
Ok Sid thank you for what you have done  i deeply appreciate it..

I tried the command you say,"From Youtube play" she goes through voice
responses but nothing opens in vlc
i tried  'from youtube play' sids e classroom again nothing opened

shivasiddharth, 2017-12-19:
So you should have got two voice responses, first  which says "fetching youtube link" and a second voice response saying "Playing Youtube Video" ? Try this command. "From Youtube play pizza recipe video"

Tony1957, 2017-12-19:
Hi Sid

I tried that command i had a few voices trying to get through in the
background
and one saying i cant play videos i haven't got a screen

shivasiddharth, 2017-12-19:
Fair enough... I need some time, i shall look into it and get back.

Tony1957, 2017-12-19:
Ok thank you, i'm off to bed now.

shivasiddharth, 2017-12-20:
just remembered.. videos dont play, when they are run in the background. That is an inherent thing with all the headless systems. To be able to see the video, you should run the main.py manually

Tony1957, 2017-12-20:
Hi Sid
Good Morning

I just tried that i got sound up for the video but no picture, i took a
copy of what i got
pi@raspberrypi:~ $ /home/pi/GassistPi/src/main.py
-bash: /home/pi/GassistPi/src/main.py: Permission denied
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ sudo /home/pi/GassistPi/src/main.py
sudo: /home/pi/GassistPi/src/main.py: command not found
(env) pi@raspberrypi:~ $ /home/pi/GassistPi/src/main.py
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': ''}
ON_CONVERSATION_TURN_TIMEOUT
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'from YouTube play'}

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 168, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 161, in main
    say("Fetching YouTube links for, "+track)
  File "/home/pi/GassistPi/src/actions.py", line 95, in say
    os.remove(tempfile)
FileNotFoundError: [Errno 2] No such file or directory: 'temp.wav'
(env) pi@raspberrypi:~ $ /home/pi/GassistPi/src/main.py
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'from YouTube play homemade pizza recipe'}
homemade pizza recipe
VLC media player 2.2.7 Umbrella (revision 2.2.7-0-g6e32381286)
[01b64170] pulse audio output error: PulseAudio server connection failure:
Connection refused
[01b68eb8] core interface error: no suitable interface module
[01ad48f8] core libvlc error: interface "globalhotkeys,none" initialization
failed
[01b68f20] dbus interface error: Failed to connect to the D-Bus session
daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[01b68f20] core interface error: no suitable interface module
[01ad48f8] core libvlc error: interface "dbus,none" initialization failed
[01ad48f8] core libvlc: Running vlc with the default interface. Use 'cvlc'
to use vlc without interface.
[01b694e0] qt4 interface error: Could not connect to X server
[01b694e0] skins2 interface error: cannot initialize OSFactory
[01b694e0] [cli] lua interface: Listening on host "*console".
VLC media player 2.2.7 Umbrella
Command Line Interface initialized. Type `help' for help.
> [01ae4120] core playlist: stopping playback
[74301180] xcb vout display error: window not available
[74301180] xcb vout display error: window not available
[74301180] xcb vout display error: window not available
[74301180] xcb vout display error: window not available
commandline read: vlc
commandline read: https://www.youtube.com/watch?v=Lyj8wH2iTYg

   ~~~~~~~~~~~~~~~~~~~~~~~~~~| DirectFB 1.2.10 |~~~~~~~~~~~~~~~~~~~~~~~~~~
        (c) 2001-2008  The world wide DirectFB Open Source Community
        (c) 2000-2004  Convergence (integrated media) GmbH
      ----------------------------------------------------------------

(*) DirectFB/Core: Single Application Core. (2014-11-01 05:58)
(!) DirectFB/core/vt: Error opening `/dev/tty0'!
    --> Permission denied
(!) DirectFB/Core: Could not initialize 'system_core' core!
    --> Initialization error!
[74301180] directfb vout display error: Cannot create primary surface
[74301180] fb vout display error: cannot get terminal mode (Inappropriate
ioctl for device)
72068950] avcodec decoder error: more than 5 seconds of late video ->
dropping frame (computer too slow ?)
                          [72068950] avcodec decoder error: more than 5
seconds of late video -> dropping frame (computer too slow ?)
                                                     [72068950] avcodec
decoder error: more than 5 seconds of late video -> dropping frame
(computer too slow ?)

   [72068950] avcodec decoder error: more than 5 seconds of late video ->
dropping frame (computer too slow ?)
                          [72068950] avcodec decoder error: more than 5
seconds of late video -> dropping frame (computer too slow ?)
                                                     [72068950] avcodec
decoder error: more than 5 seconds of late video -> dropping frame
(computer too slow ?)

   [72068950] avcodec decoder error: more than 5 seconds of late video ->
dropping frame (computer too slow ?)
                          [72068950] avcodec decoder error: more than 5
seconds of late video -> dropping frame (computer too slow ?)
                                                     [72068950] avcodec
decoder error: more than 5 seconds of late video -> dropping frame
(computer too slow ?)

   [72068950] avcodec decoder error: more than 5 seconds of late video ->
dropping frame (computer too slow ?)

On Wed, Dec 20, 2017 at 10:31 AM, shivasiddharth <notifications@github.com>
wrote:

> just remembered.. videos dont play, when they are run in the background.
> That is an inherent thing with all the headless systems. To be able to see
> the video, you should run the main.py manually
>
> —
> You are receiving this because you were mentioned.
> Reply to this email directly, view it on GitHub
> <https://github.com/shivasiddharth/GassistPi/issues/80#issuecomment-353026675>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AgpPGOe5qDO0IrTNtIgtGuJ4e0f7TIy0ks5tCOIcgaJpZM4Q0M_I>
> .
>

-- 

http://porttalbotbouncycastleshire.com

Tony1957, 2017-12-20:
Hi Sid

I tried that with putty

I just tried it again in the desktop of pi terminal and it brought up the
vlc console and flashed an image of the pizza video and the sound plays, i
think it is just a matter of some settings somewhere.

But it's nearly there.

shivasiddharth, 2017-12-20:
Its due to the lack of software codes for decoding video. I can suggest just you one thing at this point, may be you can get a Pi Zero W, install OSMC on it and leave it plugged to the TV or Monitor.

Tony1957, 2017-12-20:
Ok thank you Sid
i'll try that

shivasiddharth, 2017-12-20:
Ok, I am closing this issue for now, if you have any more concerns, please feel free to reopen it again..

closed 2017-12-20 by shivasiddharth

Tony1957, 2017-12-20:
ok thank you for your patience Sid

Tony1957, 2017-12-21:
Hi Sid

I have installed osmc on another pi2 set it all up i now ot the youtube
videos playing great,
i am struggling a bit with the music playing, and shuffling.

I noticed in the main.py and assistant.py there are some lines that
say Uncomment the following after starting the Kodi, have they all got to
be uncommented in both files.

Regards
Tony

shivasiddharth, 2017-12-21:
If you are using the custom wakeword then they need to be uncommented in assistant.py else if you are using just the hey google, then main.py would suffice. If you uncomment those lines, kodi would be muted, when GA starts listening to you and audio would be restored after the end of the conversation.

Tony1957, 2017-12-21:
Ok thank you Sid.

Regards
Tony

Tony1957, 2017-12-21:
Hi Sid Just one more question and i think we have cracked it,
is there a way to increase the listening time when we say ok google?

Regards
Tony

shivasiddharth, 2017-12-21:
Unfortunately, we dont have control over it.

Tony1957, 2017-12-21:
Ok just a thought

Tony1957, 2017-12-22:
Hi Sid

Sorry to trouble you again,
I am having issues with testing out your script, i have tried a few fresh
installs and i am getting the same issue with credentials.

I tried with a creating a new project set the credentials correctly, and i
got the same issue
any idea's.

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ google-assistant-demo
usage: google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
                             --device_model_id DEVICE_MODEL_ID
                             [--project_id PROJECT_ID]
google-assistant-demo: error: the following arguments are required:
--device_model_id
(env) pi@raspberrypi:~ $

shivasiddharth, 2017-12-22:
Google updated their SDK on 20th with new features. The change was radical and has broken all the existing projects, incl AIY's official ones.. I am almost done porting the project with the new sdk, installation process has changed a bit, so will update the git and release an How-To video tonight..

noahwater, 2018-01-15:
Hi Sid,
Im getting this error:
#

pi@NoahPi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id noahpi-19xx01  --device_model_id noahpi-19xx01-20180115111410
usage: main.py [-h] [--credentials OAUTH2_CREDENTIALS_FILE] --device_model_id
               DEVICE_MODEL_ID [--project_id PROJECT_ID]
main.py: error: argument --credentials: No such file or directory: "/home/pi/.config/google-oauthlib-tool/credentials.json"