shivasiddharth/GassistPi #81: Unable to Add Skills to Alexa Error The Language you selected doesn't match Alexa Account Settings

akishoreplanon, 2017-12-05:
Hi Sid, 

I have selected the Language en-EU while configuring the Alexa. When i try to add the skill to my Alexa Pi from web/app. I see the error The Language you selected doesn't match Alexa Account Settings.

For the following source https://github.com/alexa/avs-device-sdk/wiki/Raspberry-Pi-Quick-Start-Guide.
I came to know that en-IN is also supported. Can you please let me know if we can change the language from en-EU to en-IN. 

Thanks
Kishore

shivasiddharth, 2017-12-05:
Sorry, you are on the wrong git !

closed 2017-12-05 by shivasiddharth

akishoreplanon, 2017-12-05:
Hi Sid, Yes Is it was en-GB not en-EU it was a typo. Why can't we change the language to en-IN ?

shivasiddharth, 2017-12-05:
I know that Alexa has been released in India, but for language selection, no such option exists.. If you want an answer for why it does not exist, you need to contact the Alexa guys.