shivasiddharth/GassistPi #85: Snowboy custom wack up word is work slowly

kumarmohan7, 2017-12-11:
Hi,
Snowboy custom wack up word is work very slowly,
Ok Google is work good, 
Snowboy is sum times work, sumtime not work,
one time work  try next time not working...
Please tell how to solve this issues....

Thanks.

shivasiddharth, 2017-12-12:
What do you mean by slow working ?

kumarmohan7, 2017-12-13:
Hi,
Thanks for Reply,

My Custom Wack Word 1st time work good then i try again wack up is not working, but "Ok Google" is working good i try many times its working,
But My Wack Up word is not working after ling time i try my wack up word its work...

that is the problem,   

please help how to solve this issues,

Thanks.

kumarmohan7, 2017-12-13:
Hi,
Thanks for Reply,

My Custom Wack Word 1st time work good then i try again wack up is not working, but "Ok Google" is working good i try many times its working,
But My Wack Up word is not working after long time i try my wack up word sumtimes its work...

that is the problem,

please help how to solve this issues,

Thanks.

shivasiddharth, 2017-12-13:
Unfortunately, there is nothing that can be done. While recording/creating the snowboy model, you need to check for the sensitivity. After updating the project to Python3, i solved all of the snowboy/ GA grpc issues. Triggering is more to do with model sensitivity that you are using. For more details on the sensitivity, please check out the snowboy documentation.

kumarmohan7, 2017-12-13:
Hi,
Thank u for Reply,

I install GassistPi 4 days Before using your Git,
In your latest GassistPi is Python 3 updated or not or how to check ?
After Create snowboy i test in snowboy website its working good,
Here how to check sensitivity...?

Thanks,

shivasiddharth, 2017-12-15:
I updated the project to Python3 almost one month back, so what you are running must be the python3 version. It is always advisable to train the snowboy model in a silent environment and the same mic that you use with pi. After recording the model, you will be asked to test the model. There you should test the model for different sensitivities. If sensitivity is set higher, then you will get false positive detections, or if sensitivity is low, you may not get detections at all. So always, the model must be checked before using with pi.

kumarmohan7, 2017-12-17:
Hw to Run GassistPi in Python3
I install & using latest only 
any changes i will do 
please guide to me 

Thanks,

shivasiddharth, 2017-12-19:
We can keep going in circles all day along with this.. I have already tld you that what you are running is in Python3, still you are asking me the same thing again.  I have again, given you the reason for your issue and have even suggested as to what needs to be done.

closed 2017-12-19 by shivasiddharth