shivasiddharth/GassistPi #86: gassistpi-ok-google.service not initializing properly at startup

worldwidewoogie, 2017-12-12:
gassistpi-ok-google.service was not initializing properly at startup because the network was not fully initialized.  The following updated service file fixed the issue for me:

[Unit]
Description=Google Assistant
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py
WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

shivasiddharth, 2017-12-12:
Thanks for the shout., i have modified your post and shall leave it open just in case.

shivasiddharth, 2017-12-22:
I have updated the systemd files in the git. Closing this issue now. Thanks for your insight.

closed 2017-12-22 by shivasiddharth