shivasiddharth/GassistPi #88: Installer Issues

lucidwan, 2017-12-16:
First - thanks a ton for the work on this. I've been needing something more - and something to mess around with.

I've had a few issues with the install routine on 2 separate - clean installs on newly imaged SD card using Stretch.  apt upgrade and kernel upgrade first.
Wheel builds fail on the install. To fix this - I installed the google-assistant-sdk first - rerun both install scripts. 
After this - the installer is writing the google-oauth-tools/credentials.json and the mps-youtube folder to /root/.config. 
This throws an error on upon starting the script in the env. This latest Stretch build forces a "sudo su" to access /root/* .
So you must use  "sudo su" and then  cp -r /root/.config/* /home/pi/.config.

After this I was able to get it all working.

Thanks again - time to play.

Eldhen, 2017-12-18:
Had the exact same problem with wheel build (no problems with the oath or youtube paths thou). Followed your advice and manage to get it to work. Thank you!

shivasiddharth, 2017-12-19:
@lucidwan , Thanks for the shout. Have changed the title and will leave it open for anyone else, who might run into this issue.

suyashpaliwal, 2018-08-29:
i tried `sudo su 
cp -r /root/.config/* /home/pi/.config` using this terminal command which prompted me an error ::'cp: cannot stat '/root/.config/*' :no such file or directory .

I even tried manually creating a google-oauthlib-tool folder in .config folder and copying credentials.json file in it whcih prompted me an error:: no such option: --project-idalexabot-175e4

I am using raspberry pi 1b model.

unsigneduk, 2018-11-23:
I tried this solution on my raspberry pi zero W, but stills reports the same missing file after starting manually. Using Debian 9.