shivasiddharth/GassistPi #89: Respeaker 2 Mic Hat

wangodanny, 2017-12-16:
Hello

I have a Respeaker 2 Mic Hat: https://www.amazon.co.uk/ZIYUN-ReSpeaker-dual-microphone-expansion-applications/dp/B074Z25HXX/ref=sr_1_1?ie=UTF8&qid=1513442978&sr=8-1&keywords=ZIYUN+ReSpeaker+2-Mics+Pi+HAT

I don't know how to make this hat work with your project and some help would be appreciated. I have already set this up but using with GassistPi is hard. Thanks

shivasiddharth, 2017-12-19:
Google Assistant has audio issues with all of those Respeaker HATs. They do not work out of the box.   I definitely, would not recommend any of the Respeaker HATs for google assistant project. That said, i will try my best to assist you. Try running Pulseaudio in the background and also try changing the number of channels in the asound.conf from 2 to 1.

valascus, 2017-12-23:
I got respeaker 4 mic hat working fine. It took a lot of time, but i did it.
Actually, i've wiped my microsd card to update to the new version and i'm right now doing it again.

wangodanny, 2017-12-26:
Oh, that's great. Please could you post how you did that as it would be
really helpful. Thanks

noemu, 2017-12-31:
i've got it working with an diffrent Project.
Maybe it helps.
i'm using the driver from seeeed [https://github.com/respeaker/seeed-voicecard](url)

I've also deactivated snd_bcm2853 (modify ```dtparam=audio=on``` to ```#dtparam=audio=on``` in /boot/config.txt )
and modified the alsa config (/etc/asound.conf)
That's for the 2mic version
```
pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
    }

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave {
        pcm "hw:0,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
        format S16_LE
}
    ipc_key 1024
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:0,0"
        channels 2
    }
    ipc_key 666666
}
```

I'm not very familiar with alsa and i'm working on a pi zero w.

jabulonboaz, 2018-01-31:
@wangodanny did you try the solution described by @noemu or did you solve the issue any other way? 
My Pi zero w and ReSpeaker 2mic arrive this week I will attempt @noemu solution.

@valascus how did you manage to solve it exactly?

wangodanny, 2018-02-01:
I used that solution as i couldn't find any other way elsewhere. Just get
the install the drivers and then change the id. Drivers:
 https://github.com/respeaker/seeed-voicecard
<https://github.com/respeaker/seeed-voicecard>. Then note the id that the
pi gives you check the devices and change them in asoundrc to match them.

jabulonboaz, 2018-02-02:
@wangodanny I get this
**** List of PLAYBACK Hardware Devices ****
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ 

Were do i replace and what values ?
This what is in "asoundrc"
options snd_rpi_googlemihat_soundcard index=0

pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

jabulonboaz, 2018-02-02:
i also tried to change asoundrc and asound conf and put in both what @noemu has inserted but its not working.
I can hear the assistant initiating and saying hello and sound does come through but i speak and it does not react it looks like the microphones are not working

jabulonboaz, 2018-02-02:
i also get this when i try to record
pi@raspberrypi:~/seeed-voicecard $ arecord -f cd -Dhw:1 | aplay -Dhw:1
ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) ALSA lib pcm_hw.c:1713:(_snd_pcm_hw_open) Invalid value for card
aplay: main:788: Invalid value for card
arecord: main:788: audio open error: No such file or directory
audio open error: No such file or directory
pi@raspberrypi:~/seeed-voicecard $

jabulonboaz, 2018-02-03:
I have fixed this by modding the asoundrc and asoundrc.cofig

BadFlo, 2018-02-04:
Hi, can you upload your modified asoundrc.config file toaybe add it to the project?

jabulonboaz, 2018-02-04:
Ok will do soon as i am done trying to make it work with pi zero w now. I got only one respeaker hat and took it off pi3 and trying to install ga on pi0w now make it hotword activated.

jabulonboaz, 2018-02-09:
Note: i do not know how to submit this as a contribution to this project so i wrote it here i hope that @Sid will use this in a coments section for others to benefit from and thank uou Sid!
I have successfully instaled  using AssistantsPi Alexa on a Pi zero W using the Pi3 instal process and it works with wakeword activation - no need for the button.
I have not only completed the install on a Pi3 but allso on a PiOw and it works like a charm. I have only installed Alexa from the options of assistant pi as with Google assistant and Alexa running simultaneously it won't work on Pi0 but it does on Pi3 btw.
The way i done it was to install the AssitantsPi by choosing option 2 Alexa only and choose 1 for AIY hat as soundcard option. That leaves the asound.conf empty. I have an AIY hat so i could test the sound but u don't need to have one i believe you can just go ahead and finish the install any way without confirming the sound check. Then when i finished installing on the pi3 (btw i skipped the very last command of the instructions on the Assistant Pi git that activated Google assistant) i removed the AIY hat and placed the Respeaker 2 mic on the Pi3. I Then downloaded the Rspeaker 2mic drivers from their git https://github.com/respeaker/seeed-voicecard and the rebooted.
Miraculously the respeaker drivers had deleted the only other appearing audio card that was the AIY hat's and the asound.conf file was populated. BTW Copy pasting the contents of the asound to asoundrc made no difference.
Alexa is working perfectly the only thing there is no "hello" sound when she starts up it must be because i skipped the last command that was supposed to activate the Google assistant i assume it contained some .wav for Alexa or maybe simply the instal script runs in a way that if u install only alexa you get no "hello" at startup. Any way i then took the very SD card with that Pi3 build and another miracle happened when i put it and the respeaker on a Pi zero W it works there too and i have sound from all channels! Only thing on the Pi0 it takes a good 3 to 5 minutes to boot up as the Pi0 is slow.
Any way very please with Alexa from AssistantsPi works headless and with the hotword activation -no button needed, and the package of the device is so small it can fit into anything. My next step it to use Snowboy for custom wake words and activate some of the GPiO pins to be able to control a relay from the PiO. Thank you Sid!

Pnether, 2018-02-09:
I use a Seeed Respeaker2-mic without problems by following the instructions
as were given in a magazine to build a DIY Google Assistant with the respeaker.
The installation did not give problems and works well.

RASPBIAN VERSION as installed prior to Github GassistPi installation:
---------------------------------------------------------------------
	2017-11-29-raspbian-stretch (Kernel 4.9)

PUT RESPEAKER BOARD ON THE RASPBERRY:
-------------------------------------
	sudo apt install git         (note: latest update was already on pi)
	git clone https://github.com/respeaker/seeed-voicecard.git
	cd seeed-voicecard
	sudo ./install.sh 2mic

REBOOT 1:
---------
	sudo reboot                  

CHECK 1: LIST OF ALL AUDIO DEVICES:
-----------------------------------
a) check list with various recording devices: arecord -l
	pi@raspberrypi:~ $ arecord -l
	  **** List of CAPTURE Hardware Devices ****
	  card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  	  Subdevices: 1/1
  	  Subdevice #0: subdevice #0
	pi@raspberrypi:~ $ 
b) check list with various playback apparaten: aplay -l
	pi@raspberrypi:~ $ aplay -l
	  **** List of PLAYBACK Hardware Devices ****
  	  card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
	    Subdevices: 8/8
	    Subdevice #0: subdevice #0
	    Subdevice #1: subdevice #1
	    Subdevice #2: subdevice #2
	    Subdevice #3: subdevice #3
	    Subdevice #4: subdevice #4
	    Subdevice #5: subdevice #5
	    Subdevice #6: subdevice #6
	    Subdevice #7: subdevice #7
	  card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
	    Subdevices: 1/1
	    Subdevice #0: subdevice #0
	  card 1: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
	    Subdevices: 1/1
	    Subdevice #0: subdevice #0
	pi@raspberrypi:~ $ 

AUDIO DEVICES CONFIGURATION:
----------------------------
Switch off the onboard audio of the Raspberry, by creating a NEW file
with contents "blacklist snd_bcm2835" (one line only, without the quotes):
	cd /etc/modprobe.d
	ls
	sudo nano /etc/modprobe.d/blacklist-alsa.conf       
             (  insert a line with:    blacklist snd_bcm2835  )

REBOOT 2:
---------
	sudo reboot 

CHECK 2: LIST OF 2-MIC AUDIO DEVICE ONLY:
-----------------------------------------
a) check list arecord -l        (compare with check 1 above: same, only seeed hardware)
	pi@raspberrypi:~ $ arecord -l
	  **** List of CAPTURE Hardware Devices ****
	  card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  	    Subdevices: 1/1
  	    Subdevice #0: subdevice #0
        pi@raspberrypi:~ $
b) check list aplay -l           (compare with check 1 above, only seeed hardware)
	pi@raspberrypi:~ $ aplay -l
	  **** List of PLAYBACK Hardware Devices ****
	  card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
	    Subdevices: 1/1
	    Subdevice #0: subdevice #0
	pi@raspberrypi:~ $ 

VERIFY 1 SOUND RECORDING AND PLAYBACK:
--------------------------------------
        arecord -f S16_LE -d 5 test.wav         (record sound for 5 seconds)
	aplay test.wav				(playback works, with a lot of background noise,
                                                   but that seems no problem for Gassist).
CONFIGURE ALSA MIXER SETTINGS:
------------------------------
        pi@raspberrypi:~ $ alsamixer
        in F3 Playback use the L en R arrow keys to select a device / bar.
		Set the headphones on level 0 
                and set the two speakers on the right on level 0.
        in F4 Capture use the L en R arrow keys to select a device / bar.
		Set Capture on level 84.
        	<esc> = exit
        in F5 All the corrections are only visible after <esc> and restart of AlsaMixer.

VERIFY 2 SOUND RECORDING AND PLAYBACK:
--------------------------------------	
        arecord -f S16_LE -d 5 test.wav         (record sound for 5 seconds)
	aplay test.wav				(playback works, with a lot of background noise)
                                                   but that seems no problem for Gassist).

VERIFY 3 SOUND PLAYBACK:
------------------------
	speaker-test -t wav                      (as in Gassist installation guide)

REBOOT 3: TO BE SURE EVERYTHING IS OK
-------------------------------------
	sudo reboot 

Now the Google Assistant can be installed, as described on weblink https://github.com/shivasiddharth/GassistPi. 
proceed with alinea "CONTINUE after SETTING UP AUDIO".

jabulonboaz, 2018-02-10:
This is a tip on how to make the new Respeaker drivers to work provided to me by a member of their staff as I had issues with their new driver installation. The new driver is suppose to fix any issues we had with the typical git cloning process. This is the instruction I wore for another trundled fella that had issues like me to try help him:

‘@cogmeta i am waiting for @Pillar1989 to to clarify it as when he did it it was via team viewer and he told me he will provide full instructions, He did it too fast so i could not follow everything but its like a few commands only, whilst you wait here is the relevant "history" of cmd form my pi that gives a clue as to what he did and by following the bellow i belive you will succeed:
(I have added some comments #IN CAPITAL LETTERS for you next to the commands for you @cogmeta, line 27 to 34 is when he took over and before that i was just following his lead - the commands are bellow use them without the numbers and the comments):

8 git clone https://github.com/respeaker/seeed-voicecard -b install
9 cd seeed-voicecardsudo ./install.sh
10 pi@raspberrypi:~ $ #IGNORE THIS
11 cd seeed-voicecard
12 sudo ./install.sh
13 sudo reboot
14 aplay -l #AT THIS POINT THE DRIVER HAD INSTALLED BUT THE RE-SPEAKER CARD WAS NOT APPEARING - THIS WAS THE PROBLEM IN THE FIRST PLACE
15 jabulonboaz @jabulonboaz 09:32 #IGNORE THIS
16 i2cdetect -y 1 #I THINK THIS WAS NOT WORKING
17 i2cdetect -y #I THINK THIS WAS NOT WORKING
18 i2cdetect -y 1 #I THINK THIS WAS NOT WORKING
19 sudo apt install i2c-tools
20 i2cdetect -y 1 #I THINK HERE THE COMMAND WAS " i2cdetect -y* " with * not "1" BUT TRY BOTH CANT REMEMBER IF THIS COMMAND WAS WORKING
21 cat /boot/config.txt
22 cat /boot/config.txt | grep i2c
23 ls /dev/i2c* #THIS WAS NOT WORKING
24 ls /dev/i2c #THIS WAS NOT WORKING
25 ls /dev/i2c* #THIS WAS NOT WORKING
26 ls /dev/ # THIS COMMAND WORKED -PILLAR WAS IN THE TERMINAL NOW VIA TEAM VIEWER
27 sudo raspi-config #HERE I BELIEVE HE ENABLED I2C IN THE RPI MENU UNDER INTERFACES
28 ls /dev/i2c-1
29 cat /boot/config.txt | grep i2c
30 sudo i2cdetect -y 1
31 sudo systemctl restart seeed-voicecard
32 aplay -l #SUCCESS THE RE-SPEAKER CARD NOW APPEARS !!!
33 arecord -l #SUCCESS THE RE-SPEAKER CARD NOW APPEARS !!!
34 arecord -f cd | aplay #THIS COMAND IS TESING THE MIC AND SPEAKER DO NOT WORRY ABOUT THE ECHO SOUND THIS IS BECAUSE THE MIC IS TOO SENSITIVE IF CLOSE TO A SPEAKER THERE IS ECHO, AFTER YOU HEAR IT HIT CONTROL+C

@cogmeta bear in mind that we did this with a new clean install of a Raspbian Jessie lite and installing the respeaker 2mic drivers was the first step before i proceeded with the installation of AlexaPi . Now before i started installing AlexaPi i blacklisted the RPi internal sound card just in case, like this:
#Create this file like this:
cd /etc/modprobe.d
sudo nano alsa-blacklist.conf
#PUT THIS LINE IN THE FILE
blacklist snd_bcm2835
sudo reboot

Then when you come back to the terminal you do "aplay -l" and you are only left with the respeaker 2mic voice card. When i completed the AlexaPi installation its was working straight away i never need to do anything in the audio settings of the AlexaPi configuration file. I did made one change though and that was to activate GPIO 17 on the AlexaPi configuration file for the repseaker button to work and be able to activate Alexa from there too as well as my voice - this you can find in the final part of the instructions given in the respeaker git by @Pillar1989 in this link
https://github.com/respeaker/seeed-voicecard/wiki/AlexaPi
‘

shivasiddharth, 2018-02-11:
@jabulonboaz , I had to delete your comment, because it was filled with either IGNORE THIS or NOT WORKING and the instructions you posted was outdated. @Pnether 's post seems to be the latest/working instruction.

jabulonboaz, 2018-02-11:
@shivasiddharth this was not the same, this was how to install the new drivers the above post shows how to install the old drivers. Both work for AssitantsPi though but in some cases the old method does not. I had it all explained but its up to you. 
The comments in capital letters where there as part of the "history" explanation in fact i took the time to explain everything that the Seeed engineer did to fix the issue by displaying tje entire history. Instead of deleting the entire post you could have optionally deleted these parts then but hey your git your call. 
Thank you any way i love assistanspi.

shivasiddharth, 2018-03-28:
Closing this for inactivity.

closed 2018-03-28 by shivasiddharth