shivasiddharth/GassistPi #91: Assistant stuck in INFO:root:Playing assistant response.

dudukk1312, 2017-12-18:
Hi,

Today I have problem when use this project.
I try googlesamples-assistant-pushtotalk to test Assistant but when I tell something and Assistant response, it's stuck in `INFO:root:Playing assistant response.` and didn't have any response for me. I tried snowboy.py too. Voice trigger work fine but Google Assistant always stuck. I installed project in Raspbian, my hardware is Raspberry Pi B+ and just follow the guideline, installation is fine without error.

> (env) pi@raspberrypi:~ $ googlesamples-assistant-pushtotalk
> INFO:root:Connecting to embeddedassistant.googleapis.com
> Press Enter to send a new request...
> INFO:root:Recording audio request.
> WARNING:root:SoundDeviceStream read overflow (3200, 6400)
> INFO:root:End of audio request detected
> INFO:root:Finished playing assistant response.
> Press Enter to send a new request...
> INFO:root:Recording audio request.
> INFO:root:End of audio request detected
> INFO:root:Transcript of user request: "what's do I have today".
> INFO:root:Playing assistant response.

Thank you

shivasiddharth, 2017-12-19:
@dudukk1312 , For quite some time, there has been some issues with the gRPC API. Please read through this https://developers.google.com/assistant/sdk/release-notes. I am not sure, whether the assistant guys have fixed it. You need to check this up with those guys

shivasiddharth, 2017-12-22:
Project has migrated to the new SDK update. The old gRPC issues have been solved by google. Closing this issue..

closed 2017-12-22 by shivasiddharth