shivasiddharth/GassistPi #94: google-assistant-demo error

sohumr123, 2017-12-20:
followed the instructions very carefully and I was able to get into the (env) and then the command
google-assistant-demo gives me an error that reads:

env) pi@raspberrypi:~ $ google-assistant-demousage: google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
                             --device_model_id DEVICE_MODEL_ID
                             [--project_id PROJECT_ID]
google-assistant-demo: error: the following arguments are required: --device_model_id

shivasiddharth, 2017-12-20:
Are you here from Youtube and which Pi are you using ?

sohumr123, 2017-12-20:
yes im from youtube saw your videos and im using the Aiy kit board and Pi 3

shivasiddharth, 2017-12-20:
A quick question. Are you using the AIY image or Raspbian ?

sohumr123, 2017-12-20:
i think that's the problem im just seconds away from trying it with the the raspbian  image .ill let you know

sohumr123, 2017-12-20:
Still no luck getting the same error and yes this time im using raspbian . not Aiy Image

shivasiddharth, 2017-12-20:
Google has just released an update to make pi detectable on home app. That has broken this project. I need to update this project according to the new release, untill then this project cannot be used.

richardvdweerd, 2017-12-22:
Hi,

I just installed the custom voice hat and configured a raspberry following the instruction video (Rpi3, Raspbian, Sid's custom voice hat).
I ran into the same problem as sohumr123: getting the same error on running the google-assistant-demo.

I can get it run adding an argument: google-assistant-demo --device_model_id=googlevoicehat-soundcard. The assistant works fine and responses correctly.

After reboot (as instructed in the video) the assistant keeps silent. Running manual with the argument then it works fine.

Digging into it I found an answer of Sid addressing another problem to run: python /home/pi/GassistPi/src/main.py.

That gave an error also: missing 1 required positional argument: 'device_model_id'

My python knowledge is not great at this moment, but I tried to add an argument to the main() function in main.py script:
parser.add_argument('--device_model_id', metavar='DEVICE_MODEL_ID', default='googlevoicehat-soundcard')

Unfortunately it didn't help. I assume that the issued parameters are wrong. But I hope that adding this argument is pointing in the right direction.

What to do next?

shivasiddharth, 2017-12-22:
@richardvdweerd , I have not updated the project to the new sdk yet. Work is almost done, will run some final checks and update the git later today.

richardvdweerd, 2017-12-22:
Thanks! You're great. I will try it, as soon as the git is updated.

shivasiddharth, 2017-12-22:
@sohumr123 , @richardvdweerd  Update the project to work with the updated SDK. Installation process has slightly changed, if you follow the instructions in the README, you should have the headless GA up and running. Closing this issue for now. If you have some issue with the updated git, please feel free to reopen this issue.

closed 2017-12-22 by shivasiddharth

Enzoinventor, 2017-12-23:
is this command still to be updated?
python3 /home/pi/GassistPi/src/pushbutton.py