shivasiddharth/GassistPi #95: 100 % processor use all the time.

Priyajit08, 2017-12-20:
I am using a raspberry pi zero W
recently after updating to the 29th Dec 17 update and running the snowboy.service the process use is always 100% earlier it was less than 50% usage.
Please tell me what I should do?

shivasiddharth, 2017-12-20:
I think the Kodi functions is causing this revert back to the old project then. use:
`git clone https://github.com/shivasiddharth/GassistPi -b f2d5028416c7ab4c536d655f2e1cef2502d7e957`
This will clone the old repo.

Priyajit08, 2017-12-21:
Thanks for the solution, can I selectively disable the Kodi functions in the latest build, Instead of an revert back?

shivasiddharth, 2017-12-21:
In the actions.py i have marked the start and end of Kodi functions. You can disable them and also remove those from imports in assistant.py

closed 2017-12-21 by Priyajit08