shivasiddharth/GassistPi #97: missing 1 required positional argument: 'device_model_id'

bmericc, 2017-12-21:
main.py not start

python /home/pi/GassistPi/src/main.py
> Traceback (most recent call last):
>   File "/home/pi/GassistPi/src/main.py", line 152, in <module>
>     main()
>   File "/home/pi/GassistPi/src/main.py", line 122, in main
>     with Assistant(credentials) as assistant:
> TypeError: __init__() missing 1 required positional argument: 'device_model_id'
>

shivasiddharth, 2017-12-21:
Google Updated their SDK yest. I need to make the modifications to match that. Work is in progress. Will update the Git tomorrow

Enzoinventor, 2017-12-21:
it does not work yet. even I myself problem. it works only if I run:
google-assistant-demo --project_id "*****" --device_model_id "*****"

shivasiddharth, 2017-12-21:
@Enzoinventor Lke i said, yest update broke the project, but work is almost done here is the new branch (https://github.com/shivasiddharth/GassistPi/tree/newsdk). Some final round of tweaking left and tomorrow GassistPi users can have a headless unit running on latest SDK

Enzoinventor, 2017-12-21:
@shivasiddharth ok thank you so much we look forward to your updates, we did not even have time to try the new kody feature you've just created !!

Enzoinventor, 2017-12-22:
@shivasiddharth very kind you should update the link because it still clones the old script:
FIRST STEP- CLONE the PROJECT on to Pi
Open the terminal and execute the following
git clone https://github.com/shivasiddharth/GassistPi

shivasiddharth, 2017-12-22:
I have updated the Git, If you follow the instructions in the readme, headless GA should be up and running. Closing this issue, now, if you have some issues with the updated git, please feel free to reopen.

closed 2017-12-22 by shivasiddharth

Enzoinventor, 2017-12-23:
sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh
/home/pi/GassistPi/scripts/gassist-installer-pi3.sh: riga 36: /home/pi/modelid.txt: permission denied
why?

shivasiddharth, 2017-12-23:
Did you run it without sudo command ? You need to run it with sudo

Enzoinventor, 2017-12-23:
@shivasiddharth I ran it with sudo and I also gave a chmod + x to modelfile.txt.
Now seeing that I get too many errors I format the sd and reinstall the image aiyprojects-2017-09-11 and clone your project again.
you confirm that I must also update sdk otherwise it will not work properly?
Thank you very much for your patience

shivasiddharth, 2017-12-23:
@Enzoinventor , This is not AIY proj. I have clealy mentioned in the README that you should not use AIY image with this project.

Enzoinventor, 2017-12-23:
ok,   
I installed raspbian desktop, following the guide from scratch I get this error is normal?
Downloading protobuf-3.5.1-py2.py3-none-any.whl (388kB)
    100% |████████████████████████████████| 389kB 463kB/s 
Collecting pycparser (from CFFI>=1.0->sounddevice<0.4,>=0.3.7; extra == "samples"->google-assistant-sdk[samples])
  Downloading https://www.piwheels.hostedpi.com/simple/pycparser/pycparser-2.18-py2.py3-none-any.whl (209kB)
    100% |████████████████████████████████| 215kB 772kB/s 
Requirement already up-to-date: idna<2.7,>=2.5 in ./env/lib/python3.5/site-packages (from requests>=2.0.0->requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples])
Requirement already up-to-date: chardet<3.1.0,>=3.0.2 in ./env/lib/python3.5/site-packages (from requests>=2.0.0->requests-oauthlib>=0.7.0->google-auth-oauthlib[tool]>=0.1.0->google-assistant-sdk[samples])
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (from protobuf>=3.0.0->googleapis-common-protos>=1.5.2->google-assistant-grpc==0.1.0; extra == "samples"->google-assistant-sdk[samples])
Building wheels for collected packages: grpcio
  Running setup.py bdist_wheel for grpcio ... error
  Complete output from command /home/pi/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-odxru9e3/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpekmxld32pip-wheel- --python-tag cp35:
  Found cython-generated files...
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio

Enzoinventor, 2017-12-23:
ok the installation went to fine and everything works finally. thank you so much